FreeCAD/FreeCAD #13873: [Problem] Spreadsheet table is not correctly aligned

NXTler, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello all,
As you can see in this screenshot, the rows and columns are slightly shifted:
[image]
It's nothing critical and should have the lowest priority, but it still adds to the unfinished feel of Spreadsheet.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git) AppImage
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * PieMenu 1.2.7
  * sheetmetal 0.4.13
  * A2plus 0.4.64a
  * QuickMeasure 2022.10.28
  * Curves 0.6.36
  * CfdOF 1.25.11
  * IconThemes
  * fasteners 0.5.20
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-07:
What theme are you using? It's not as bad on my side but still visible if you know it:
[image]

NXTler, 2024-05-07:
The theme I use is "Dark modern".

luzpaz, 2024-08-14:
@MisterMakerNL can you reproduce ?

xtemp09, 2024-12-07:
Is this issue still relevant?

tarman3, 2024-12-07:
[image]

<details>
  <summary>FreeCAD 1.1.0dev.39416 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.0; Silk 0.1.5; Curves 0.6.51
```
</details>

MisterMakerNL, 2024-12-07:
@tarman3 Seems to be an offset of a pixel on the corner cell!
I also think the cell it is the wrong color. I think we should fix this!
[image]

MisterMakerNL, 2024-12-07:
Making it look better was pretty easy fixing the offset isn't.
[image]
`/* QHeaderView ------------------------------------------------------------

https://doc.qt.io/qt-5/stylesheet-examples.html#customizing-qheaderview

--------------------------------------------------------------------------- */
QTableCornerButton::section {
  padding: -2px;
  margin: -2px;
  background-color: #333333;
  border-radius: 3px; /*if you remove this line it breaks it*/
  border: 2px solid #d70606;
}

QHeaderView {
  background-color: transparent;
  /* text-align: bottom; */
}

QHeaderView:disabled {
  /* background-color: #444444; */
  color: #adadad;

}

QHeaderView::section {
  background-color: qlineargradient(x1:0, y1:0.3, x2:0, y2:1, stop:0 #333333, stop:1 #252525);
  color: #ffffff;
  padding: 4px;
}

QHeaderView::section::horizontal {
  border-bottom: 1px solid #020202;
  border-right: 0px solid transparent;
  border-top: 1px solid #020202;
  border-left: 1px solid #020202;
}

QHeaderView::section::horizontal::first, QHeaderView::section::horizontal::only-one {
  /* border-left: 0px solid #020202; */
}

QHeaderView::section::horizontal:disabled {
  color: #adadad;
}

QHeaderView::section::vertical {
  border-bottom: 0px solid transparent;
  border-right: 1px solid #020202;
  border-top: 1px solid #020202;
  border-left: 1px solid #020202;
  padding-left: 4px;
  padding-right: 4px;
}

QHeaderView::section::vertical::first, QHeaderView::section::vertical::only-one {
  /* border-top: 2px solid #020202; */
}

QHeaderView::section::vertical:disabled {
  border-top: 2px solid #020202;
}
`

MisterMakerNL, 2024-12-25:
I couldn't find this topic anymore tnx Max :)

closed 2025-01-06 by yorikvanhavre