FreeCAD/FreeCAD #13874: Spreadsheet cell color buttons don't reflect state of selected cell

krushia, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The formatting buttons in the spreadsheet toolbar don't reflect the state of the selected cell. They instead show the last format applied to any cell in any spreadsheet open. This confuses the user when for example they have a cell with fg/bg white/black but the buttons show black/white. In this case I get tricked into using the wrong button to change a color.

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37100 (Git) AppImage
Build type: Release
Branch: main
Hash: ede6c811fd39d68a44949601f2eb7ecc8a67fdb7
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36
  * A2plus 0.4.64a
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * Alternate_OpenSCAD 1.0.0
  * CurvedShapes 1.0.9
  * fasteners 0.5.20
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * workfeature
  * MnesarcoUtils 0.2.5
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-07:
I guess this is to easily apply the last setting to other cells. This should need some more discussion.

monkeypangit, 2024-05-07:
I think there are two simultaneous issues working in tandem here:

1. The issue described by the krushia
2. The fact that cells are highlighted by filling in the background of the cell and changing the text color based on style sheet settings (effectively masking its selected forward and background color).

It would probably make sense to do the following:

1. Toolbar button colors reflect the colors of the currently selected cell.
2. If multiple cells are selected the colors are set to that of the first selected cell (QEvent::MouseButtonPress).
3. Draw a frame around selected cells and background color as currently (this seems to be a style sheet setting for the QTTableWidget)

This matches the behavior of for example Google Sheets.

How to set the QTTableWidget stylesheet:
https://www.qtcentre.org/threads/27195-Border-around-selected-cell-in-qtablewidget

luzpaz, 2024-07-12:
Anyone willing to open a forum thread and link to it from this ticket ?

xtemp09, 2025-01-10:
> The formatting buttons in the spreadsheet toolbar don't reflect the state of the selected cell.

I have checked _Word_. The colors in the toolbar reflect the last color applied, not the color of the cell. So, the behavior of _Spreadsheet_ follows the standard behavior of other programs.