FreeCAD/FreeCAD #13875: Start WB: make "Parametric Part" Button, create a Part with a Body in it

tobiasfalk, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, the Button "Parametric Part" creates a Body, I think it would be intelligent to put this Body in to a Part. Some reasons for it:

1. It says "Part" and not Body
2. Doing it "forces" some structure on to the user, may seam dumb, but it would benefit the user in the long run.
3. A user who wants a pure body in the file can still use the "Empty file" Button

An other thing that would be nice if the user is asked how the Part should be named before creation, so that the user does not need to rename the Part/Body

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly4
  * fasteners 0.4.59
  * GDML
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-05-07:
Duplicate of #12938.

adrianinsaval, 2024-05-07:
closed in favor of the above

closed 2024-05-07 by adrianinsaval

tobiasfalk, 2024-05-07:
Sorty did not find that one🤦‍♂️