FreeCAD/FreeCAD #13876: api-ms-win-core-path-l1-1-0.dll missing when executing Windows portable distribution

TwoRedCells, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When attempting to execute the 64-bit Windows portable distribution, upon unzipping the directory structure and running `bin/freecad.exe`, I get *System Error* _The program can't start because api-ms-win-core-path-l1-1-0.dll is missing from your computer..._

I have tried starting execution from the main directory and from the `bin` directory.

* Windows 7 Professional SP1
* Intel i7-2600K, 16GB RAM

### Full version info

```shell
Unable to open application.
```

### Subproject(s) affected?

None

### Anything else?

I assume the installer will work, but wanted to report this before I try the installer, since that may make the issue unreproducable in the future.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-05-07:
I assume when you state Portable Distribution you mean  _FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310.7z_  ?
If so, then you'll notice it's built with Python 3.10.x which is not compatible with Windows 7 SP1. You'll have to use the proper installer which has Python 3.8.x from https://github.com/FreeCAD/FreeCAD/releases

kaiwas, 2024-05-07:
The problem is partially solved with this recipe.
https://github.com/nalexandru/api-ms-win-core-path-HACK
This is due to the fact that the required version of Python is not supported by Windows7. But this recipe worked before Python 10.
I again cannot run assemblies based on version 11.
But freecad version 0.21.1 definitely works (0.21.2 seems to have problems)

TwoRedCells, 2024-05-07:
Thank you. Perhaps the download page could be updated to reflect this, as the text "Windows 7 is the minimum supported version. For more info on installation, please check out the Wiki" is directly under the download link for "64-bit portable (.7z)", and the linked Wiki page does not mention this constraint.

luzpaz, 2024-08-14:
Indeed as of today the download page does indeed say this: 

[image]

How should we proceed ?