FreeCAD/FreeCAD #13877: [CAM]  Base Geometry widget tries to delete original face from the job instead of removing it from the list

phaseloop, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**Steps to reproduce:**
- create VCarve operation in CAM workbench (probably this issue manifests itself with other operations as they share UI code)
- add face to "Base geometry" widget
- select added face name on the list and hit "delete" on keyboard

**Current behaviour:**
"Object dependency" alert is raised that deleting this face may break this operation (and others which reference this particular face). Hitting "yes" will remove whole object (this face belongs to) from the job "Model" section.

**Expected behaviour:**
hitting "delete" button will remove face from the "Base geometry" list as "Remove" button below the widget does.

### Full version info

```shell
0.22dev - revision 14555
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Russ4262, 2024-08-11:
Is there documentation to support the intended behavior of the 'DEL' key when using the 'Base Geometry' widget?  This behavior may not be a bug. The behavior might be a feature request instead if no documentation indicates the intended behavior of the 'DEL' key within this widget.

Respectfully,
Russell

phaseloop, 2025-03-28:
Necromancing this thread. There is absolutely no reason why any object should be deleted from a job when hitting del key inside operation base window. There is a delete button doing right thing - removing object from the list so natural expectation is that del key on keyboard does the same.

closed 2025-07-10 by github-actions[bot]