FreeCAD/FreeCAD #13882: Bundle build scripts should be more easily found

yorikvanhavre, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As a follow-up of #13871, the location of the bundle build scripts and their documentation should be indicated in the FreeCAD README file to be more easily found by users

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-05-07:
I will prepare a document describing the build steps and then we can reference that either in the main readme or in a readme in tools/build

yorikvanhavre, 2024-05-08:
There is already some docs at https://freecad.github.io/DevelopersHandbook/technical/ReleasePackages.html

luzpaz, 2024-07-18:
Any traction on this ?