FreeCAD/FreeCAD #13883: Sketcher: No OVP/Widget for B-spline tools

FEA-eng, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, the OnView Parameters and tool settings widget are not used for the polyline (#12462) and arc of ellipse/hyperbola/parabola (#12463) tools. But they could potentially be used for the B-spline tools as well - just to enter the coordinates of their points.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-05-07:
duplicate of #12016?

FEA-eng, 2024-05-07:
> duplicate of #12016?

Apparently. I haven't noticed that one as the search returned only the issues created by Max (likely because the issue created by Paddle has the spelling "Bspline" which is not found when looking for "spline" or "B-spline").

closed 2024-05-07 by FEA-eng

maxwxyz, 2024-05-07:
edited the other issue's title, thanks!