FreeCAD/FreeCAD #13890: Task view width unbounded

krushia, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I double-clicked on a Part Fusion in the Tree view, and the Task view exploded in size.
[image]

In the same session I was using sketcher, and the task view didn't behave oddly there. A list of things I've done in the past day that could maybe have caused it:

1. I've changed from FreeCAD_weekly-builds-37100-conda-Linux-x86_64-py311.AppImage to FreeCAD_weekly-builds-37213-conda-Linux-x86_64-py311.AppImage
2. I've made some tweaks to system fonts via fontconfig to diagnose other issues
3. I've moved Tree view and Selection view from outside the main window to inside, then back again a few times to diagnose yet another reported issue

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git) AppImage
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36
  * A2plus 0.4.64a
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * Alternate_OpenSCAD 1.0.0
  * CurvedShapes 1.0.9
  * fasteners 0.5.20
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * workfeature
  * MnesarcoUtils 0.2.5
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

I've seen this happen in other Qt apps. KDE has one seldom used dialog that has MAX_INT horizontal dimension. I can "fix" it with a window manager rule. The screenshot was taken after moving Tasks out of the window (it was unbounded when docked as well). This is a 4 monitor setup, one on top of 3.

Related: https://github.com/FreeCAD/FreeCAD/issues/11016
Possibly caused by: https://github.com/FreeCAD/FreeCAD/pull/13611 (which I see has been reverted)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-05-08:
Caused by using the Terminus font. There is a longstanding Qt bug about this. https://bugreports.qt.io/browse/QTBUG-22599

I'm leaving this open because even though Qt is bugging out, the problem only manifests when no maximum size is defined for windows and/or the widgets contained within.

krushia, 2024-05-09:
Related: https://github.com/FreeCAD/FreeCAD/issues/10514

FEA-eng, 2025-10-02:
@krushia Can you check if this issue is still relevant?

luzpaz, 2026-09-03:
> Caused by using the Terminus font. There is a longstanding Qt bug about this. "Polish characters of Terminus font not rendered properly" https://bugreports.qt.io/browse/QTBUG-22599

So what do we do ?

luzpaz, 2026-10-12:
> It's an old bug, but still open on the Qt side, so it may still be relevant.

@FEA-eng the bug was closed 1 month ago as `Not_in_maintenance_scope`  
My previous question still relevant: https://github.com/FreeCAD/FreeCAD/issues/13890#issuecomment-3691340560