FreeCAD/FreeCAD #13893: Blank 3D display when running under Kwin Wayland

nikola825, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm running FreeCAD under KDE6 with Wayland.
When I try to create a new project or open an existing one, the 3D view is just blank
[image]

I see the following output in the terminal when the blank screen is shown
Tried enabling the software renderer, it didn't help
```
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

The `.FCStd#` backup format is deprecated as of v0.21 and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
Coin warning in cc_glglue_instance(): Error when setting up the GL context. This can happen if there is no current context, or if the context has been set up incorrectly.
Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
Coin warning in SoGLPolygonOffsetElement::updategl(): OpenGL driver doesn't support z-buffer offsetting
```

But when I run it in X11 mode by calling
`QT_QPA_PLATFORM=xcb freecad`
Everything works fine:
[image]

I'm running the package from the Arch linux repositories
It's running under Kwin window manager in Wayland mode
I have an Nvidia GPU with the latest drivers

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Anything else?

Here's the file I used in the screenshots. But it probably doesn't matter, even starting a new empty project gives the same problem
[cube.zip](https://github.com/FreeCAD/FreeCAD/files/15242160/cube.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-05-08:
Known issue in Coin3D. Many distros patch FreeCAD to use xcb. https://github.com/coin3d/coin/issues/451

adrianinsaval, 2024-11-18:
given that there's better info at #17491 I'll close this in favor of that one

closed 2024-11-18 by adrianinsaval