FreeCAD/FreeCAD #13895: BIM - Icon updates task

marcuspollio, 2024-05-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following the [unification](https://github.com/FreeCAD/FreeCAD/pull/13783) of the Arch, BIM and NativeIFC workbenches, some work on the icons is needed. This issue gathers them on one place.

### Update icons for a consistent style

* [ ] TODO

### Add new icons for [IFC elements](https://standards.buildingsmart.org/IFC/RELEASE/IFC4_3/HTML/lexical/IfcBuiltElement.htm)

* [ ]  IfcBuildingElementProxy
* [ ]  IfcCovering
* [x]  IfcBeam
* [x]  IfcColumn
* [x]  IfcCurtainWall
* [x]  IfcDoor
* [ ]  IfcMember
* [ ]  IfcRailing
* [ ]  IfcRamp
* [ ]  IfcRampFlight - Question: do we really need a dedicated icon for that ? Ramp flights are aggregated in IfcRamp, whatever the amount of flights and landings (IfcSlab).
* [x]  IfcWall
* [x]  IfcSlab
* [ ]  IfcStairFlight - Same question as RampFlight
* [x]  IfcWindow
* [x]  IfcStair
* [x]  IfcRoof
* [ ]  IfcPile
* [ ]  IfcFooting
* [ ]  IfcBuildingElementComponent
* [ ]  IfcPlate

### Full version info

```shell
OS: GNU/Linux Ubuntu 22.04 LTS based (DE:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 +11 (Git)
Build type: Release
Branch: pr/13783
Hash: 98844db62dd45d53abdfddfaec2df04b199de637
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.5.1
```

### Subproject(s) affected?

Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marcuspollio, 2024-05-10:
For the first item (update icons with consistent "document" style), the left ones are the present icons, the right ones the proposed :

[image]

Feedback welcome.

yorikvanhavre, 2024-05-13:
LGTM!

marcuspollio, 2024-05-13:
> LGTM!

Cool ! I'll prepare a PR once [your big merge](https://github.com/FreeCAD/FreeCAD/pull/13783) is committed.
I'll have a look at the other IFC elements soon.

maxwxyz, 2024-05-19:
not sure where to put this but it's not worth a new issue, maybe this typo can be fixed with a UI update:
[image]

maxwxyz, 2024-05-19:
Same with this typo in BIM:
[image]

maxwxyz, 2024-05-19:
also this string/function seems not necessary anymore:

marcuspollio, 2024-05-20:
See [PR 14160](https://github.com/FreeCAD/FreeCAD/pull/14160) for the first item on the list.

maxwxyz, 2024-05-20:
Looks good. I think the glue icon needs an overhaul as well. On my system it's just a red blob, not sure if intended

Roy-043, 2024-06-13:
FYI: 2 icons have been renamed as a result of #14560 2 icons have been renamed.

maxwxyz, 2024-06-29:
is the progress in the original post up to date? Anything what could be done prior to UI freeze in 2 days?

Roy-043, 2024-07-01:
According to [this forum post](https://forum.freecad.org/viewtopic.php?p=759605#p759605) the UI freeze has already started on Monday the 26th of June.

yorikvanhavre, 2024-07-01:
We [postponed it](https://forum.freecad.org/viewtopic.php?p=767646#p767646) of one week, which is today...

marcuspollio, 2024-07-03:
Hi there.
The icon updates for the 1.0 release are now done.

I propose to keep this list open but on hold until main is open again for changes for the next future release, ok ?

yorikvanhavre, 2024-07-04:
Yes!

marcuspollio, 2025-02-04:
If @yorikvanhavre don't mind, I'd like to be assigned to this issue.
@maxwxyz could you please make this change (the `1.1` milestone is already set) ? TIA

yorikvanhavre, 2025-02-04:
Done ;)

maxwxyz, 2025-02-04:
Before you start: there is a new icon guideline in  the works and @obelisk79 wanted to redo all icons.
Just so there is no work / time wasted.

obelisk79, 2025-02-04:
max is correct

marcuspollio, 2025-02-06:
As per https://github.com/FreeCAD/FreeCAD/pull/19427 I updated the TODO list here. It now includes some Draft WB stuff.

marcuspollio, 2025-03-08:
Hi @maxwxyz 
Could you please change the issue type to "task". TIA !

maxwxyz, 2025-03-08:
done

luzpaz, 2025-08-30:
What is left here to do ? Will this be ready before 1.1 release ?

closed 2025-09-30 by marcuspollio