FreeCAD/FreeCAD #13897: Fit all and Fit selection use the same icon

furgo16, 2024-05-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using the freecad-daily PPA, the icons for the <kbd>Fit all</kbd> and <kbd>Fit selection</kbd> toolbar commands are the same:

[image]

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Installed mods: 
  * OpenTheme 2024.5.3
  * parts_library
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-05-08:
You need to add the Boolean parameter  _Bitmaps/Theme/ThemeSearchPaths_ as  per https://wiki.freecad.org/Fine-tuning

furgo16, 2024-05-08:
Thanks! This works around the issue for those two icons, but I'll leave it open, as I feel it's still a bug (either in packaging or on choosing the icon, or on the system icon theme). In general, I'm happy with the system theme.

It seems that at least on Linux, there are at least 3 themes that are applied depending on the FreeCAD package installed.

| Icons | Package | Bitmaps/Theme/ThemeSearchPaths | Theme | Notes |
| --- | --- |--- | --- | --- |
|[image]<br>[image] | freecad-daily | true | Yaru | Ubuntu's default theme |
| [image]<br>[image] | freecad-daily | false | ? | Looks like Tango icons. Is this the FreeCAD icon set? Also what it looks like on default installation with the snap package |
| [image]<br>[image] | AppImage weekly | true | ? | Yet another theme. Unclear where it comes from |
| [image]<br>[image]| AppImage weekly | false | ? | Looks like Tango icons. Is this the FreeCAD icon set? |

maxwxyz, 2024-05-08:
Is it something which could be solved in FreeCAD or is there no system icon to distinguish between them?

furgo16, 2024-05-08:
I'm not sure how the translation FreeCAD icon => System icon works. FreeCAD does name them differently though (`zoom-all` and `zoom-selection`). Or if using the system theme some system icons can be overriden by FreeCAD icons.

https://github.com/FreeCAD/FreeCAD/blob/47fd87af511cc99cf0b10bb7b97169a727a5ce09/src/Gui/CommandView.cpp#L1595-L1605

https://github.com/FreeCAD/FreeCAD/blob/47fd87af511cc99cf0b10bb7b97169a727a5ce09/src/Gui/CommandView.cpp#L1563-L1574

krushia, 2024-05-09:
On my system they both look the same in the appimage. The icon they resolve to is `zoom-fit-best` from my scheme selected in KDE. There is an actual bug here, because "zoom-selection" and "zoom-all" aren't listed in the Freedesktop [Icon Naming Specification](https://specifications.freedesktop.org/icon-naming-spec/icon-naming-spec-latest.html). This means an application can't be sure they will exist. See also the related [Icon There Spec](https://specifications.freedesktop.org/icon-theme-spec/icon-theme-spec-latest.html)

I have no idea how this works on Windows. The [QIcon](https://doc.qt.io/qt-5/qicon.html) docs are a long read.

krushia, 2024-05-09:
I spoke too soon. FreeCAD does have those named icons in the source,
https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/Icons/zoom-fit-best.svg
https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/Icons/zoom-all.svg
... but apparently they don't get used if Bitmaps/Theme/ThemeSearchPaths is false? In my case there is no setting defined. 
[image]

furgo16, 2024-05-09:
> ... but apparently they don't get used if Bitmaps/Theme/ThemeSearchPaths is true? In my case there is no setting defined.

That was the same behavior on my system. Before I created the property, the two icons distributed with FreeCAD were not used, so I assume if `Bitmaps/Theme/ThemeSearchPath` is not defined, it defaults to 'true' => use system theme icons. As per the [table](https://github.com/FreeCAD/FreeCAD/issues/13897#issuecomment-2099967437).

If I understand it correctly,

1) from https://doc.qt.io/qt-5/qicon.html#fromTheme, and at least on Linux: that method is used to fetch the system icon given a descriptive string (e.g. `zoom-all`). 

https://github.com/FreeCAD/FreeCAD/blob/82101ac89049eea6e89a9c57c83ccf5282a3c03c/src/Gui/Command.cpp#L981-L989
_Just as an example, there are other places where that method is called_

Then also from the docs:

> If an icon can't be found in the current theme, then it will be searched in [fallbackSearchPaths](https://doc.qt.io/qt-5/qicon.html#fallbackSearchPaths)() as an unthemed icon.

2) The [Icon Naming spec](https://specifications.freedesktop.org/icon-naming-spec/icon-naming-spec-latest.html) defines `zoom-fit-best`, `zoom-in`, `zoom-original` and `zoom-out`.  Neither of those map to the `zoom-all` or `zoom-selection` strings defined in the FreeCAD code to load those icons.

3) Despite not being a match, `zoom-fit-best` is the system theme icon returned when calling `iconFromTheme()` for either `zoom-original` or `zoom-out`.

This would mean that his bug is about finding out why 3) is happening.

furgo16, 2024-10-05:
Related: https://invent.kde.org/foresteam/breeze-icons/-/commit/ff66b122c49a4ce33567c78c8367a0f68834c0a4

furgo16, 2024-10-05:
Related bug: https://bugs.kde.org/show_bug.cgi?id=450870 - this seems to indicate that the AppImage uses the [Breeze icon set](https://invent.kde.org/frameworks/breeze-icons), which is the default KDE Plasma icon set.

furgo16, 2024-10-05:
Related FreeCAD issue: https://github.com/FreeCAD/FreeCAD/issues/6633