FreeCAD/FreeCAD #13899: A problem about FreeCADcmd to import dxf format file

sigurHeBin, 2024-05-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have successfully compiled and installed the FreeCAD command line tool from source code in linux.
```
[root@iZn11er29gnfasZ qt5]# FreeCADCmd --version
FreeCAD 0.21.2 Revision: 33771 (Git)
``` 
I want to import a file in dxf format through a python script and then export it to another format.
This is my python script:
```
# -*- coding: utf-8 -*-
import FreeCAD
import Draft
import Mesh

def import_dxf_and_export_stl(input_path, output_path):
    
    doc = FreeCAD.newDocument('MyModel')

    
    Draft.importDXF(input_path)
    
    Mesh.export(doc.Objects, output_path)

    
    doc.close()

input_path = "/app/software/blender-python/bzz.dxf"  
output_path = "/app/software/blender-python/bzz.stl"  

import_dxf_and_export_stl(input_path, output_path)
``` 
I ran the script, but I got an error
```
[root@iZn11er29gnfasZ qt5]# FreeCADCmd /app/software/blender-python/export_dxf_to_stl2.py
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
(c) Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Exception while processing file: /app/software/blender-python/export_dxf_to_stl2.py [module 'Draft' has no attribute 'importDXF']

``` 

I have printed all the attributes and methods available in the Draft module, but I did not find any import methods, but I found in the document that it was imported through the Draft module. 
May I ask how to import the dxf format model by using the command line tool？
Looking forward to your reply！thanks！

### Full version info

```shell
[root@iZn11er29gnfasZ FreeCAD]# FreeCADCmd --version
FreeCAD 0.21.2 Revision: 33771 (Git)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-05-08:
@sigurHeBin for help using FreeCAD, you'll get better support using the FreeCAD forum at https://forum.freecad.org/viewforum.php?f=3. GitHub is for reporting issues that need to be fixed, or for feature requests. I'd suggest you post your question to the forum. I hope this helps!

Syres916, 2024-05-08:
Change the top of your code to:

```
# -*- coding: utf-8 -*-
import FreeCAD
import Draft
import importDXF
import Mesh

def import_dxf_and_export_stl(input_path, output_path):
    
    doc = FreeCAD.newDocument('MyModel')

    
    importDXF(input_path)
```

maxwxyz, 2024-05-08:
I'll close this. Help request on the forums please.

closed 2024-05-08 by maxwxyz