FreeCAD/FreeCAD #13904: UI: Overlay settings show no icon

maxwxyz, 2024-05-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It seems that Dock window overlay > Toggle left, right, top, bottom ... show no icon but instead an X, maybe for a missing file?
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.12
  * Curves 0.6.36
  * dodo 1.0.1
  * fasteners 0.5.20
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.3
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-15:
What are the icons for these functions anyway?

xtemp09, 2024-09-15:
@maxwxyz, https://github.com/FreeCAD/FreeCAD/blob/b4923d1c584a4dce794d91a4b090a5293569223a/src/Gui/CommandView.cpp#L3706-L3716

luzpaz, 2024-09-17:
Looks like the icons are '[correct](https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/Stylesheets/overlay/icons/close.svg)' meaning it renders the icon that was intended. I guess we need to decide what these icons should actually be?

closed 2025-06-20 by kadet1090