FreeCAD/FreeCAD #13905: Can't deactivate or delete constraints in Sketcher WB

mfraser, 2024-05-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using FreeCAD 0.22~pre1~202405071449~ubuntu22.04.1, I am unable to delete an existing constraint in the sketcher WB. If I hi-light the constraint I want to delete, nothing happens and as soon as I move off the constraint it is unselected. I can't even delete them from the constraint list.

This is a repeat of https://github.com/FreeCAD/FreeCAD/issues/13169

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * QuickMeasure 2022.10.28
  * fasteners.backup1712686361.5869133 0.5.16 (Disabled)
  * kicadStepUpMod 11.1.4
  * fasteners 0.5.20
  * lattice2.backup1713204440.8139822 1.0.0 (Disabled)
  * Curves 0.6.36
  * ThreadProfile 1.89.0
  * sheetmetal 0.4.13
  * VendorParts
  * lattice2 1.0.0
  * kicadStepUpMod.backup1714893889.389688 11.1.3 (Disabled)
  * Freecad-Built-in-themes-beta 1.2.2
  * Lithophane
  * ProDarkThemePreferencePack 1.0.0
  * Curves.backup1714929565.0217447 0.6.35 (Disabled)
  * CurvedShapes.backup1713382123.4246979 1.0.8 (Disabled)
  * Manipulator 1.5.7
  * Defeaturing.backup1713382129.3425534 1.2.2 (Disabled)
  * DynamicData 2.62.0
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.9
  * Assembly4 0.12.4
  * freecad.gears 1.2.0
  * A2plus 0.4.64a
  * fcgear 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-09:
@bgbsww FYI

mfraser, 2024-05-09:
Seems to be fixed as of this version.
```
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37241 (Git)
Build type: Release
Branch: main
Hash: 3bb6f92885021c5a5667c728486eaf6f8d515524
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * QuickMeasure 2022.10.28
  * fasteners.backup1712686361.5869133 0.5.16 (Disabled)
  * kicadStepUpMod 11.1.5
  * fasteners 0.5.20
  * lattice2.backup1713204440.8139822 1.0.0 (Disabled)
  * Curves 0.6.36
  * ThreadProfile 1.89.0
  * sheetmetal 0.4.13
  * VendorParts
  * lattice2 1.0.0
  * kicadStepUpMod.backup1714893889.389688 11.1.3 (Disabled)
  * Freecad-Built-in-themes-beta 1.2.2
  * Lithophane
  * ProDarkThemePreferencePack 1.0.0
  * Curves.backup1714929565.0217447 0.6.35 (Disabled)
  * CurvedShapes.backup1713382123.4246979 1.0.8 (Disabled)
  * Manipulator 1.5.7
  * Defeaturing.backup1713382129.3425534 1.2.2 (Disabled)
  * DynamicData 2.62.0
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.9
  * Assembly4 0.12.4
  * freecad.gears 1.2.0
  * A2plus 0.4.64a
  * fcgear 1.0.0
```

closed 2024-05-09 by mfraser