FreeCAD/FreeCAD #13910: [Problem] V0.22dev Sketcher: CreatePoint causes crash of FreeCAD

7th-fret, 2024-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If in a sketch there is a geometry point, constrained or not, upon exit of sketch causes crash of FreeCAD.
This happens only with geometry and not with construction point.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37238 (Git)
Build type: Release
Branch: main
Hash: 82101ac89049eea6e89a9c57c83ccf5282a3c03c
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-05-09:
I can't confirm this on Windows.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
```

maxwxyz, 2024-05-09:
@PaddleStroke / @bgbsww FYI

bgbsww, 2024-05-14:
Thanks, I'll get to it.  A few items to complete first.

PR #14045 (2024-05-16): Toponaming: Fix point crash bug #13910
Fix #13910 by echoing existing code that avoids setting an element name on standalone geometry points
closed 2024-05-16 by chennes