FreeCAD/FreeCAD #13911: [Feature Request/Structural change] "Splitting" of TD

tobiasfalk, 2024-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I think it would be beneficial and good if, the TD WB would be "split". What I mean by it is that the Base drawing preparations, like creating a page, inserting a Template, exporting a Page and insert Annotation should be available for creating another WB that also creates a Page but does not need the ability to insert a View of a 3D body and insert dimension. 
By splitting TD in to a Base Draw WB and Tech Draw (witch inherits Base Draw) it would allow other WBs to use it, some that come to my mind are a Schematic WB ([posted in the Forum about it once](https://forum.freecad.org/viewtopic.php?t=73654)) and a BOM WB could use it to create a Paged BOM for printing. This would also automatically prevent the existing of multiple Template formats for a Page, since every WB base of this uses the same.

I understand that this is not something that will happen tomorrow or even before 1.0, but I still think that doing it would make future development easier and the Program more stream lined.
After the Split is done, one can look if it would be possible to make this base also available for the creation of Python based WBs.

@WandererFan there are definitely drawbacks to this, but I do not know enough about TD that I am able to name them, a´do you know them?

To recap:

Splitting TD in to:
- Base Draw, a WB base that has the ability to create a page, inserting a Template, exporting a Page, insert Annotation and other basic drawing stuff
- Tech Draw, to the user the same as now, but it inherits Base Draw, and so its ability, and then adds all the other stuff it has like inserting a View and adding measurements

Benefits:

- Gives the ability to use this base to create other WB that need a Page and so on
- Prevents multiple WB exclusive Page Templates, since they all use the same Page base

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * DynamicData 2.60.0
  * fasteners 0.5.15
  * kicadStepUpMod 11.1.0
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tobiasfalk, 2024-05-09:
@FEA-eng 
I would not actually make this "WB" visible to the user, since it is more of a building block of a WB than a whole WB it self. 
Maybe me calling it a WB was misleading.
And splitting because as to how I see it, the code for the Page creation and so would be "removed" from TD and moved in to its one "quasi/sude" WB that is then inherited by TD, and so from outside there is no change, since the Tool are still visible and usable to the User. But when one than creates another WB with that also inherits this "quasi/sude" WB than the user also sees the base tool for creating a page and so on, currently the only way would be by inheriting TD, but the one needs to remove all the TD specific tool that are not needed in his WB manual, since they are never inherited.

Mannshoch, 2024-08-30:
For TD I had another thinking. From Solidworks and Solidworks MBD I know the way of doing the main work like measuring and place surface or weldment annotations in 3D and after that create paper views. (I exclude the MBD way of creating interactive PDF here)

If I take this thinking as baseline I would split TD into a _TD Annotator_ and _TD Paper_ 

In this case May TD Annotator store Information on the 3D-Model may and could may add annotations into linked sub-assembly or parts and TD Paper allows to manage the views and what annotation is seen inside.

> Base Draw, a WB base that has the ability to create a page, inserting a Template, exporting a Page, insert Annotation and other basic drawing stuff.

I like the Idea using .svg but I do not like the Idea using Inkscape. I think there is a need for an own tool. SVG could be still the way for a further Enhancement of a template.

I'm not sure what tools need to be grouped together but I think may a Template designer would be a god Idea. It's not really intuitive how I setup the page-size, create a border and title block (with attributes) and preset it with printer or PDF output settings. 
... And replace it because I think paper size A3 it not enough, I need A2.

tobiasfalk, 2024-11-26:
Hi, now that FC1 is released, I want to "ping" this and hear some more opinions if this is something reasonable.

An other "solution"/way would be to add these "basic tools" for bage creation and so, to FC Core this way every WB could use them.