FreeCAD/FreeCAD #13913: [Feature Request] Grouping of pages and Automatically adjusting things like page number

tobiasfalk, 2024-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Having multiple TD Pages is not something uncommon, and putting them into an Object Group helps with organizing them.
It is also not uncomment that one Group of pages does not really have anything to do with the other and should for example not be exported together or pages counted separately in the total sheet number for the autofill (love it).
That is why, a way of creating a Group of pages (Booklet?) where pages are counted separately or one can export all pages in it at once are available. This would also give the ability to add feature like creating a table of contents or other Page group related stuff.

Ideally, one could also link a page of one Group of pages in to another (even from a Part to an Assembly), and it still displays the correct page number and so.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * DynamicData 2.60.0
  * fasteners 0.5.15
  * kicadStepUpMod 11.1.0
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-05-09:
This is definitely a need. Assemblies often have two drawings: an assembly drawing with exploded views, and a customer drawing with external dimensions and interfaces. They are completely different so they shouldn’t share page numbers.

This problem wouldn’t exist if files had only one root (part, assembly, drawing). See discussion #11742.

WandererFan, 2024-11-02:
Page number is problematic since not all pages in a typical document will be created in FC.   There will be pages for text, spreadsheets, diagrams, cartoons, TOC, bibliography, etc.  See the comments in  #16955.

tobiasfalk, 2024-11-02:
> Page number is problematic since not all pages in a typical document will be created in FC.   There will be pages for text, spreadsheets, diagrams, cartoons, TOC, bibliography, etc.  See the comments in  #16955.

If I remember correctly, ISO 7200 differentiates between Page Nr. and Sheet Nr.

- Sheet Nr. is the index in a group of sheets, for example, with an assembly drawing one sheet for the general assembly drawing and the second one for the exploded view. Another example could be if one has a complex part and wants to have the section views split up on many different sheets.

> 5.1.6 Segment/sheet number The segment/sheet number identifies the segment or sheet. NOTE As needed, the contents of a document can be divided into fixed portions, called segments. In the case of technical drawings, these segments are called sheets. 

- Page Nr. is the page index in an overreaching document, a Manuell, for example.

> 5.3.9 Page number The page number is usually generated by the presentation system. 

Yes, this can lead to the case where there are multiple Sheets 1 in one document, but all of these sheets have different Page Nrs. and most likely different titles.

Sheet Nr. is mandatory, while Page Nr. is not.

tobiasfalk, 2024-11-02:
And I may also be guilty of mixing them up in the Text at the top🤦‍♂️

pierreporte, 2024-11-02:
It would be useful if there was a Drawing container, that would contain several Pages. The template would be applied to the Drawing container and would be common to all its children pages. It would allow to have several drawings in one document, for example a drawing that show the assembly steps of an assembly, and another one that gives its overall dimensions for customers.

pierreporte, 2025-01-03:
I created (and just closed) a duplicate of this issue: #18358.

pierreporte, 2025-01-06:
@WandererFan You can answer to my comment in https://github.com/FreeCAD/FreeCAD/issues/18358#issuecomment-2565673970 here.

pierreporte, 2025-01-19:
@WandererFan what do you think of the more detailed explanation in the comment linked above?