FreeCAD/FreeCAD #13915: [Problem] Arch: crash when creating a structure object

wwmayer, 2024-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Switch to Arch wb
2. Create a structure object
3. Click into the 3D view
4. After a while FreeCAD crashes

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37239 (Git)
Build type: debug
Branch: main
Hash: 03a371d6e37f4ce47f5ffed7d5b039c5866a7e05
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Arch

### Anything else?

Forum thread: https://forum.freecad.org/viewtopic.php?t=87493

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-05-09:
I ought to add, I cannot reproduce the crash all the time. For example, if I set radial gradient as background, the crash becomes unreproducible.

<details><summary>Backtrace</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36992 (Git)
Build type: Debug
Branch: main
Hash: 53020def864fea94b172c5d7e49614f576790023
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

```cpp
Process 1795 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
    frame #0: 0x00007ffff2a76057 libCoin.so.80c`SoAction::traverse(SoNode*) + 39
libCoin.so.80c`SoAction::traverse:
->  0x7ffff2a76057 <+39>: movq   (%rsi), %rax
    0x7ffff2a7605a <+42>: callq  *(%rax)
    0x7ffff2a7605c <+44>: movl   %eax, %edi
    0x7ffff2a7605e <+46>: callq  0x7ffff2c623c0            ; SoNode::getActionMethodIndex(SoType)
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
  * frame #0: 0x00007ffff2a76057 libCoin.so.80c`SoAction::traverse(SoNode*) + 39
    frame #1: 0x00007ffff2c10908 libCoin.so.80c`SoChildList::traverse(SoAction*, int, int) + 360
    frame #2: 0x00007ffff2c57660 libCoin.so.80c`SoGroup::doAction(SoAction*) + 80
    frame #3: 0x00007ffff2c6c66c libCoin.so.80c`SoSeparator::doAction(SoAction*) + 44
    frame #4: 0x00007ffff6f52491 libFreeCADGui.so`Gui::SoFCUnifiedSelection::handleEvent(this=0x00005555580345f0, action=0x0000555558606a90) at SoFCUnifiedSelection.cpp:750:27
    frame #5: 0x00007ffff2a76095 libCoin.so.80c`SoAction::traverse(SoNode*) + 101
    frame #6: 0x00007ffff2c10908 libCoin.so.80c`SoChildList::traverse(SoAction*, int, int) + 360
    frame #7: 0x00007ffff2c57660 libCoin.so.80c`SoGroup::doAction(SoAction*) + 80
    frame #8: 0x00007ffff2c6c66c libCoin.so.80c`SoSeparator::doAction(SoAction*) + 44
    frame #9: 0x00007ffff2a76095 libCoin.so.80c`SoAction::traverse(SoNode*) + 101
    frame #10: 0x00007ffff2a7fa9b libCoin.so.80c`SoHandleEventAction::beginTraversal(SoNode*) + 171
    frame #11: 0x00007ffff2a766e8 libCoin.so.80c`SoAction::apply(SoNode*) + 200
    frame #12: 0x00007ffff2c25559 libCoin.so.80c`SoEventManager::actuallyProcessEvent(SoEvent const*) + 121
    frame #13: 0x00007ffff6fe9636 libFreeCADGui.so`SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(this=0x00005555585fe410, event=0x0000555558606800) at QuarterWidget.cpp:1028:48
    frame #14: 0x00007ffff6ff0bba libFreeCADGui.so`SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(this=0x00005555585fe410, event=0x0000555558606800) at SoQTQuarterAdaptor.cpp:752:63
    frame #15: 0x00007ffff7025e0b libFreeCADGui.so`Gui::View3DInventorViewer::processSoEventBase(this=<unavailable>, ev=0x0000555558606800) at View3DInventorViewer.cpp:2571:37
    frame #16: 0x00007ffff6ff7a1d libFreeCADGui.so`Gui::NavigationStyle::processSoEvent(this=0x000055555860a2d0, ev=0x0000555558606800) at NavigationStyle.cpp:1486:47
    frame #17: 0x00007ffff6ffeec8 libFreeCADGui.so`Gui::CADNavigationStyle::processSoEvent(this=0x000055555860a2d0, ev=0x0000555558606800) at CADNavigationStyle.cpp:330:46
    frame #18: 0x00007ffff6ffafab libFreeCADGui.so`Gui::NavigationStyle::processEvent(this=0x000055555860a2d0, ev=0x0000555558606800) at NavigationStyle.cpp:1457:37
    frame #19: 0x00007ffff7025d83 libFreeCADGui.so`Gui::View3DInventorViewer::processSoEvent(this=0x00005555585fe410, ev=0x0000555558606800) at View3DInventorViewer.cpp:2566:36
    frame #20: 0x00007ffff6fe4bb2 libFreeCADGui.so`SIM::Coin3D::Quarter::EventFilter::eventFilter(this=0x0000555558608300, obj=<unavailable>, qevent=0x00007fffffffc3d0) at EventFilter.cpp:167:64
    frame #21: 0x00007ffff3e9ec4f libQt5Core.so.5`QCoreApplicationPrivate::sendThroughObjectEventFilters(receiver=0x00005555585fe410, event=0x00007fffffffc3d0) at qcoreapplication.cpp:1190:33
    frame #22: 0x00007ffff4ae1803 libQt5Widgets.so.5`QApplicationPrivate::notify_helper(this=<unavailable>, receiver=0x00005555585fe410, e=0x00007fffffffc3d0) at qapplication.cpp:3634:38
    frame #23: 0x00007ffff4aea60a libQt5Widgets.so.5`QApplication::notify(this=<unavailable>, receiver=0x0000555558606a00, e=0x00007fffffffc870) at qapplication.cpp:3084:43
    frame #24: 0x00007ffff6cf8206 libFreeCADGui.so`Gui::GUIApplication::notify(this=0x00007fffffffd7c0, receiver=0x0000555558606a00, event=0x00007fffffffc870) at GuiApplication.cpp:83:40
    frame #25: 0x00007ffff3e9ef19 libQt5Core.so.5`QCoreApplication::notifyInternal2(receiver=0x0000555558606a00, event=0x00007fffffffc870) at qcoreapplication.cpp:1064:24
    frame #26: 0x00007ffff3e9f1b4 libQt5Core.so.5`QCoreApplication::sendSpontaneousEvent(receiver=0x0000555558606a00, event=0x00007fffffffc870) at qcoreapplication.cpp:1474:27
    frame #27: 0x00007ffff4ae937b libQt5Widgets.so.5`QApplicationPrivate::sendMouseEvent(receiver=0x0000555558606a00, event=0x00007fffffffc870, alienWidget=0x0000555558606a00, nativeWidget=0x00007fffffffd800, buttonDown=0x00007ffff4f4d470, lastMouseReceiver=0x00007ffff4f4d450, spontaneous=true, onlyDispatchEnterLeave=false) at qapplication.cpp:2622:56
    frame #28: 0x00007ffff4b4ae1b libQt5Widgets.so.5`QWidgetWindow::handleMouseEvent(this=0x00005555570c6870, event=0x00007fffffffced0) at qwidgetwindow.cpp:684:44
    frame #29: 0x00007ffff4b4e83a libQt5Widgets.so.5`QWidgetWindow::event(this=0x00005555570c6870, event=0x00007fffffffced0) at qwidgetwindow.cpp:300:25
[... cut ...]
    frame #53: 0x00007ffff3229d90 libc.so.6`__libc_start_call_main(main=(FreeCAD`main at MainGui.cpp:105:1), argc=1, argv=0x00007fffffffdd98) at libc_start_call_main.h:58:16
    frame #54: 0x00007ffff3229e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:105:1), argc=1, argv=0x00007fffffffdd98, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd88) at libc-start.c:392:3
    frame #55: 0x0000555555569c05 FreeCAD`_start + 37
(lldb) 
```

</details>

closed 2024-05-17 by yorikvanhavre

marcosilveira2, 2024-09-22:
Choose a face on an existing object or select a preset
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7fdf5b85d090]
#1  0x7fdf57f10239 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x29
#2  0x7fdf580a74d8 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#3  0x7fdf580f25ec in SoGroup::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x5c
#4  0x7fdf5810c1ec in SoSeparator::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#5  0x7fdf5aaa1ce4 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x94
#6  0x7fdf57f10277 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x67
#7  0x7fdf580a74d8 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#8  0x7fdf580f25ec in SoGroup::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x5c
#9  0x7fdf5810c1ec in SoSeparator::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#10  0x7fdf57f10277 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x67
#11  0x7fdf57f199ab in SoHandleEventAction::beginTraversal(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xab
#12  0x7fdf57f1071a in SoAction::apply(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x28a
#13  0x7fdf580bf651 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x71
#14  0x7fdf5aa025f2 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x22
#15  0x7fdf5aa07c0b in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x5b
#16  0x7fdf5a9d388d in Gui::View3DInventorViewer::processSoEventBase(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0xd
#17  0x7fdf5a9fdbbd in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x2d
#18  0x7fdf5a9f6dff in Gui::BlenderNavigationStyle::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x25f
#19  0x7fdf5a9fe7e6 in Gui::NavigationStyle::processEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x66
#20  0x7fdf5a9d8150 in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x70
#21  0x7fdf5aa1513f in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0xaf
#22  0x7fdf588a651b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x9b
#23  0x7fdf592bea55 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x75
#24  0x7fdf592c8343 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x583
#25  0x7fdf5accc140 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x90
#26  0x7fdf588a680a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x18a
#27  0x7fdf592c7457 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#28  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c935d) [0x7fdf5931d35d]
#29  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cc1ec) [0x7fdf593201ec]
#30  0x7fdf592bea66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x86
#31  0x7fdf592c80f0 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x330
#32  0x7fdf5accc140 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x90
#33  0x7fdf588a680a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x18a
#34  0x7fdf58c8f7d3 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x6e3
#35  0x7fdf58c9110b in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x25b
#36  0x7fdf58c6b35b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xbb
#37  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x7932e) [0x7fdf5285b32e]
#38  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7fdf5693217d]
#39  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x52400) [0x7fdf56932400]
#40  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7fdf569324a3]
#41  0x7fdf588fe435 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x65
#42  0x7fdf588a53ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#43  0x7fdf588ad116 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#44  0x7fdf5ad52770 in Gui::Application::runApplication() from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x1e40
#45  freecad(+0xde54) [0x557828709e54]
#46  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7fdf5b83e083]
#47  freecad(+0xf07e) [0x55782870b07e]

yorikvanhavre, 2024-09-23:
> Choose a face on an existing object or select a preset
> Program received signal SIGSEGV, Segmentation fault.

No crash here. Could you provide more detailed step-by-step instructions of what you do to reproduce the crash?
Also, your FreeCAD info from Help -> About FreeCAD -> Copy to clipboard please?

marcosilveira2, 2024-09-23:
> No crash here. Could you provide more detailed step-by-step instructions of what you do to reproduce the crash? Also, your FreeCAD info from Help -> About FreeCAD -> Copy to clipboard please?

I had commented in the wrong place. here is the correct: https://github.com/FreeCAD/FreeCAD/issues/14535

PR #14028 (2024-05-17): Arch: Fixed trackers ending too soon
Deletes the trackers only after recompute
Fixes #13915
commit e7433585 (2024-05-17): Arch: Fixed trackers ending too soon - fixes #13915