FreeCAD/FreeCAD #13918: TechDraw: Export layered PDF

maxwxyz, 2024-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

All commercial CAD software usually export drawing in a layered PDF format. This allows the user to hide certain layers when viewing the PDF later or extract certain layers only.

Usually they provide the following layers in this order:

- pixelated image rendering (colors/shades)
- hatching
- visible lines (of the model)
- symbols (balloons, tables, finishes,...)
- section view lines (cutting lines and name)
- Cosmetic geometry (incl. page drawing frame)
- title block template
- title block content (text/images)
- dimensions (values and lines/arrows)

Demo:
[image]

With only one layer active:
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.12
  * Curves 0.6.36
  * dodo 1.0.1
  * fasteners 0.5.20
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.3
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

m1cha1, 2024-05-14:
Layers (Optional Content Groups) are supported by PDF version 1.5 (by default Freecad writes to version 1.4)
The problem is that QPdfWriter does not support layers.
However, layers are supported by [Scribus](https://github.com/scribusproject/scribus/blob/master/scribus/pdflib_core.cpp) . Is it possible to port it to Freecad?

tobiasfalk, 2024-05-15:
> However, layers are supported by [Scribus](https://github.com/scribusproject/scribus/blob/master/scribus/pdflib_core.cpp) . Is it possible to port it to Freecad?

Could be quite hard since TD uses the QT Painter classes for all the actuall drawing. 
This would requir to create a New QtPaint devic, witch is possible but not easy since you would also need to add much more funcunalyti to it than just painting, wich just do not exist in the Qt Painting logic.

I am not saying that it is not possible but quite hard.
For reference, just for fun I once started to look how hard it is and began a Latex Paint Device:

[My trie on a Latex Paint device, devinelty not butifull😁🤦‍♂️](https://github.com/tobiasfalk/PaintLatexQt)
[Explination on how Qt Painter works](https://doc.qt.io/qt-6/paintsystem.html)

m1cha1, 2024-05-16:
Another solution is to patch QT files:
[Modify Qt source code to implement PDF layered printing](https://zhuanlan.zhihu.com/p/363883759)

tobiasfalk, 2024-05-17:
> Another solution is to patch QT files: [Modify Qt source code to implement PDF layered printing](https://zhuanlan.zhihu.com/p/363883759)

Looks interesting and may be a way to do it. 
This would still mean to reimplement/ make a derivative of the QtPDF paint device, but this would also mean that more control over the output would be available.
This would then also allow the insertion of 3D Models in to the PDF.

Some things I found about this:
https://opensource.adobe.com/dc-acrobat-sdk-docs/pdflsdk/apireference/PD_Layer/PDOCG.html

maxwxyz, 2024-05-18:
then this would also enable https://github.com/FreeCAD/FreeCAD/issues/12843

m1cha1, 2024-05-23:
Another solution: using part of [PDF4QT](https://github.com/JakubMelka/PDF4QT) to generate pdf (Pdf4QtLibCore??)
License: LGPL 3.0
I'm not a programmer but I think we can use [QPainter](https://github.com/JakubMelka/PDF4QT/blob/master/PdfExampleGenerator/pdfexamplesgenerator.cpp)
"3D PDF support *(planned in year 2024)"