FreeCAD/FreeCAD #13922: Disabling BUILD_MATERIAL in CMake breaks FreeCAD build

phaseloop, 2024-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When disabling option BUILD_MATERIAL in CMake - during compilation following error occurs:

 ```
 ImportExportSettings.cpp
  ImportExportSettings.cpp
  mocs_compilation_Release.cpp
LINK : fatal error LNK1181: cannot open input file 'Materials.lib' [D:\release3\src\Mod\Part\App\Part.vcxproj]
  Automatic MOC and UIC for target E57Format
```

FreeCAD.exe binary is being built but when started - UI is non-functional (lacks all widgets and menus). 

Judging from FreeCAD compiling documentation - CMake should raise an error when one of the dependencies is not enabled.

### Full version info

```shell
0.22.0dev (main branch). Build on Windows 10 using Visual Studio 2022.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

phaseloop, 2024-05-11:
Same issue is with BUILD_ARCH - if disabled, it breaks Draft workbench which in turns breaks CAM workbench. Opening CAM Job object edit throws an error that Draft can't import arch_rc module.

luzpaz, 2024-07-18:
@davesrocketshop just wondering if you've heard or seen this before ?

davesrocketshop, 2024-07-19:
I'm not sure why this is an option. As noted, many pieces depend on materials. Disabling it is bound to create issues.

luzpaz, 2024-07-24:
@davesrocketshop then we should consider deprecating this flag then ?

davesrocketshop, 2024-07-24:
Yes.

PR #21046 (2025-05-04): Update module dependencies
Update module dependencies.

Module dependencies were incomplete. Disabling BUILD_MATERIAL resulted in failed compilation and failed unit tests as required components weren't available for various modules. This updates the dependencies so that a module can be correctly build when all the per-requisites are available, AND so that it passes all of its tests.

## Issues

fixes #13922

## Before and After Images
N/A
closed 2025-05-04 by chennes