FreeCAD/FreeCAD #13923: Sketcher generates faulty constraint when double-clicking arc

0penBrain, 2024-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce :
 * Open an empty sketch
 * Select "Arc by center" tool
 * Set the center as coincident with root point
 * Move a bit the mouse and double-click => :warning: The arc creation is discarded but a faulty coincident constraint is still created

Note 1: Steps above are simplest way to see problem, but this behavior can lead to different situation depending on how it is applied: malformed constraint, collapsing sketch, ...
Note 2: For consistency, the tool should ignore the 2nd click at same location and stay in arc creation mode

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37243 (Git)
Build type: Debug
Branch: master
Hash: 0095e52466c79c2135fce860a341554f3f51156c
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-09:
can reproduce
[image]

0penBrain, 2024-05-09:
When this one is fixed, 4499ea5b can be reverted.