FreeCAD/FreeCAD #13924: Sketcher/Polyline create null-length line if double-clicking end of tangent arc

0penBrain, 2024-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
 * Open an empty sketch
 * Select Polyline tool
 * Draw a first line then press `M` key till "tangent arc"
 * Move mouse a bit to get an arc and double-click => :warning: In better case a null-length (actually something in the 10^-6 mm range) is created and a coincident constraint added, in worst case the line is really null and it breaks solver "Fail to converge"

Would be good to prevent double-click acceptation in this case

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37243 (Git)
Build type: Debug
Branch: master
Hash: 0095e52466c79c2135fce860a341554f3f51156c
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-09:
It's not just with arc, with all modes.
What should happen on double click? End the tool?

pierreporte, 2024-05-09:
I would say end the polyline and exit tool. Currently, exiting the tool needs double right click so there would be some sort of consistency.

I don’t remember how other programs handle this though. I should check Alibre but I won’t have access to Catia until Monday.

0penBrain, 2024-05-09:
> It's not just with arc, with all modes.

I don't have it if for example I keep "broken line" mode. In this case an error is generated.

>What should happen on double click? End the tool?

No, just discard creation of a line. Eventually generate same error as in default "broken line" mode.

maxwxyz, 2024-05-09:
IMO double click should end the tool as well.

0penBrain, 2024-05-09:
> IMO double click should end the tool as well.

This isn't how other tools behave, and would be a PITA if you didn't do it on purpose. :)

maxwxyz, 2024-05-09:
how do other apps handle this? In Inkscape, GIMP, PS when creating paths or drawing pixels a double click is just used as input like currently in FreeCAD.

0penBrain, 2024-05-09:
> how do other apps handle this? In Inkscape, GIMP, PS when creating paths or drawing pixels a double click is just used as input like currently in FreeCAD.

I don't care about other apps. I care about consistency among the Sketcher tools. :wink: 
Just try with the 2nd point of a Slot and you'll see.

maxwxyz, 2024-05-09:
so you could go on: it creates a new slot if you double click on the third click of a slot or the second click on a polygon.

0penBrain, 2024-05-09:
Yes, and it starts a new one (both cases). Which is OK because in this case, the 2 clicks can be interpreted as valid points and don't create malformed geometry. :wink:

luzpaz, 2026-10-10:
Looks related to  
* #14126