FreeCAD/FreeCAD #13931: Subtractive helix looks like not working or the model is not regenerated after the feature

69Piero, 2024-05-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

in FC 0.22-37213 subtractive helix looks like not working or the model is not regenerated after the feature because no error is shown.

https://forum.freecad.org/viewtopic.php?t=87519

### Full version info

```shell
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Installed mods: 
  * 3DfindIT 1.2.0
  * Curves 0.6.36
  * dxf-library
  * freecad.gears 1.2.0
  * QuickMeasure 2022.10.28
  * symbols_library
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-10:
Is this related to: https://github.com/FreeCAD/FreeCAD/issues/12185 ?

69Piero, 2024-05-10:
no, the bug is indipendent from shape (see attacment)

[helix feature FC 0.22-37213.zip](https://github.com/FreeCAD/FreeCAD/files/15272167/helix.feature.FC.0.22-37213.zip)

FlachyJoe, 2024-05-12:
I'll take a look but this feature could be renamed HELLix :disappointed_relieved:

FlachyJoe, 2024-05-12:
The helix shape is well generated but the boolean subtraction silently fails.
You can workaround 

- with 1 turn + linear pattern but this doesn't work if Angle != 0.
- with a higher cylinder so helix ends nearer its top and then cut the excess.

It fails since #12977 merge and #13205 doesn't fix it.

FlachyJoe, 2024-06-30:
I'm not available to work on this issue until the end of July.

As I wrote, https://github.com/FreeCAD/FreeCAD/pull/12977 has to be tested. We may consider this test has failed and then we could revert the helix to the shell algo.
@bgbsww Do you have an opinion on this?

bgbsww, 2024-06-30:
Opinion?  Run screaming?  I'm a long way away from helix and its issues right now, so without going back and looking deeply, IIRC there is a historical behavior that needs to be respected, and then various more efficient alternatives.  I think the eventual solution will be to develop a working efficient alternative, and then to add the necessary property and defaults to allow the old to continue to exist but the new for all future models.  I'm not sure the difficulty of the new, although I would start super simple with tests and try to avoid adding much of any extra code.  And I'm sure it is at least a medium sized problem, certainly not as small as a single operation would have you believe.

bgbsww, 2024-08-07:
> [helix feature FC 0.22-37213.zip](https://github.com/FreeCAD/FreeCAD/files/15272167/helix.feature.FC.0.22-37213.zip)

Testing with some code, the result of the helix can be viewed on its own, any attempt at a boolean operation fails ( add / subtract both ) will fail.  If the profile or helix is modified so that it self-intersects, then everything starts working.

Or, if the helix is fully contained in the diameter of the cylinder it works - it is the intersection of the curved wall of the cylinder with the helix that triggers the failure of the boolean operations.

Doesn't matter if the cylinder is made with a padded sketch or a primitive - appears to be a fundamental OCCT refusal to work.

FlachyJoe, 2024-08-08:
@bgbsww the OCCT issue results in a missing (or disconnected) edge, see https://github.com/FreeCAD/FreeCAD/issues/15018#issuecomment-2197500203

FlachyJoe, 2024-08-25:
Fixed by #16009

PR #15678 (2024-08-05): [PD] Revert 316506f, back to shell algo for some helix cases
Fix #13931
closed 2024-08-25 by maxwxyz