FreeCAD/FreeCAD #13936: Missing buttons from task panel

davesrocketshop, 2024-05-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening a task panel, the standard buttons are missing. 

Here is what it looks like in v.21
[image]

This is what it looks like in the current weekly build. Note that the standard buttons are missing.
[image]

Steps to reproduce:
1. Select Rocket Workbench
2. Create a new document
3. Select create nose cone
4. Note missing buttons

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods:
  * CfdOF 1.25.10
  * Rocket 3.3.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-05-10:
All standard buttons appear as expected using current self compiled build:

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37246 (Git)
Build type: Release
Branch: main
Hash: 7d738f161a64b1fc154abed244603a89b3439617
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.4.13
  * Design456 0.0.1
  * Rocket 3.3.0
  * CfdOF 1.24.8
  * Assembly3 0.12.0
  * fasteners 0.5.20
  * freecad.gears 1.1.0

```

maxwxyz, 2024-05-10:
I can reproduce:
[image]
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.12
  * Curves 0.6.36
  * dodo 1.0.1
  * fasteners 0.5.20
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.3
  * OpticsWorkbench 1.0.17
  * Rocket 3.3.0
  * sheetmetal 0.4.13
```

davesrocketshop, 2024-05-10:
I was working on updating the add on to use the materials when I noticed it, but I tested it with the last release version to verify. It's definitely due to v0.22 changes. I haven't updated the release version in a while since I've been a bit busy... The WB does nothing to mess with the buttons.

It works fine when I build from source, but my repository is a little stale implying it's a recent change.

luzpaz, 2024-08-12:
Any idea what specifically is causing this  ?

davesrocketshop, 2024-08-12:
No clue. It was there through several weekly build iterations, but I'm not seeing it with the current weekly build.