FreeCAD/FreeCAD #13938: Build error, missing .h file

iplayfast, 2024-05-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

/home/chris/github/FreeCAD/src/Mod/Assembly/App/AssemblyObject.cpp:74:10: fatal error: OndselSolver/ASMTRotationLimit.h: No such file or directory
   74 | #include <OndselSolver/ASMTRotationLimit.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

### Full version info

```shell
Looking at the Onsdel version of freecad, it looks like their AssemblyObject.cpp doesn't use it.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-10:
Did you try to run
`git submodule update --init --recursive`

luzpaz, 2024-07-18:
@iplayfast :point_up_2:

iplayfast, 2024-07-24:
git submodule update --init --recursive did the struck