FreeCAD/FreeCAD #13939: Sketcher/Polyline needs 2 RMB clicks to exit after drawing a closed shape

0penBrain, 2024-05-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
 * Open an empty sketch
 * Select Polyline tool
 * Draw a closed shape (eg. a triangle) => :warning: at this time, you need 2 right clicks to exit the tool while 1 is expected

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37249 (Git)
Build type: Debug
Branch: master
Hash: 3acbc561f299b4cddea8215c387a52f3d5132cf6
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

KimK, 2024-07-15:
I am not experiencing this problem. This is with no (renamed) config files. 
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38166 (Git)
Build type: Unknown
Branch: main
Hash: 4bb6d191b4686919a2f247bcfce6c186703a3d86
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)

KimK, 2024-07-16:
@0penBrain Sorry, I made a mistake earlier here too, I did not perform the test correctly. Here it is again, and I do see the issue, screencast attached. 

https://github.com/user-attachments/assets/248f6ec2-0fb3-41c0-b07e-9667252cb031

PR #18258 (2024-12-13): Sketcher: Fix double RMB to quit polyline after closed shape
This fixes #13939 where two right mouse buttons were needed to exit the polyline tool after the creation of a closed shape.
closed 2024-12-13 by chennes