FreeCAD/FreeCAD #13940: Sketcher/Dimension tool is exited if RMB click during constraint creation

0penBrain, 2024-05-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
 * Open an empty sketch
 * Draw a Rectangle
 * Select the "Dimension" tool
 * Start creating a constraint by eg. clicking the left edge of the Rectangle
 * At this time (without validating the constraint), do a right click => :warning: The tool is completely exited, while one would expect the constraint creation is simply discarded and Dimension tool stays activated to create a new dimension

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37249 (Git)
Build type: Debug
Branch: master
Hash: 3acbc561f299b4cddea8215c387a52f3d5132cf6
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Anything else?

Ping @PaddleStroke

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

KimK, 2024-07-11:
Did someone push a fix for this? In this recent build, I am not observing this behavior, just normal behavior. 

OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38123 (Git)
Build type: Unknown
Branch: main
Hash: 342116e61480fa70bd94454a94095ed4d1db940a
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.3.13
  * fcgear 1.0.0
  * slic3r-tools
  * Lithophane
  * ProDarkThemePreferencePack 1.0.0
  * kicadStepUpMod 10.22.8
  * Assembly4 0.50.6
  * freecad.gears 1.0.0
  * fasteners 0.4.75
  * Behave-Dark-Colors 0.1.1

0penBrain, 2024-07-13:
Behavior is still present with:

KimK, 2024-07-15:
OpenBrain, I wonder if there is some difference between our two systems, as once again I do not observe this behavior. (Build updated from my previous test.) I do believe that you are seeing this issue, how can we work together to check for differences between our systems? Could it have something to do with KDE/plasma vs. Cinnamon? I'll try backing up my config file and changing it to default/vanilla config?
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38166 (Git)
Build type: Unknown
Branch: main
Hash: 4bb6d191b4686919a2f247bcfce6c186703a3d86
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.3.13
  * fcgear 1.0.0
  * slic3r-tools
  * Lithophane
  * ProDarkThemePreferencePack 1.0.0
  * kicadStepUpMod 10.22.8
  * Assembly4 0.50.6
  * freecad.gears 1.0.0
  * fasteners 0.4.75
  * Behave-Dark-Colors 0.1.1

KimK, 2024-07-15:
Tried again with no config files, still no problem seen. Any advice or suggestions?

PaddleStroke, 2024-07-15:
I can reproduce this on windows. It used to work properly so this is a regression. I will check it out.

PaddleStroke, 2024-07-15:
@maxwxyz please assign me thanks.

0penBrain, 2024-07-15:
@KimK I think there is very few chances that environment has any influence here.
Could you post a short screencast to see how your test method can differ?

closed 2024-07-15 by yorikvanhavre

KimK, 2024-07-16:
Sure, screencast attached. Did I do anything wrong? Is there something I should do differently? Please let me know. Thanks! 

https://github.com/user-attachments/assets/59933b53-6bb9-4931-af41-f2017967b5a2

0penBrain, 2024-07-16:
@KimK yes. 😄 You shall use the Dimension tool as stated: https://wiki.freecad.org/Sketcher_Dimension

KimK, 2024-07-16:
Sorry I misunderstood, my mistake. Yes, you are correct, I see the problem here too. Screencast attached. 

https://github.com/user-attachments/assets/6ee5bcd1-1808-4011-8fed-03fa90725814