FreeCAD/FreeCAD #13942: Limit values that lead to an error.

kaiwas, 2024-05-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In this example I'll mention sketcher, but in reality this behavior is present almost everywhere.
If we enter values that “break” the model or sketch, a breaking action is performed, which can be difficult to fix (it’s good if canceling the action helps).

It would be great to display a message about “valid” values when entering “breaking” data.
In the example of this sketch, the value 51 is acceptable.
at a value of 50 the sketch will break.
To avoid a breakdown, expect a pop-up message "The value cannot be less than or equal to 50"
Moreover, if the value 50 was entered, then two behaviors are possible:
1) The sketch leaves all the values ​​that were before entering the new one.
2) The value closest to the acceptable value is accepted (in this case, for example, 50.001)

Test file is attached 

[image]
[Limit_value.zip](https://github.com/FreeCAD/FreeCAD/files/15279225/Limit_value.zip)

### Full version info

```shell
0,21 & any versions
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-11:
not sure if this is possible as all values would need to be calculated.

kaiwas, 2024-05-11:
Yes. I also think that this is not an easy task.
some kind of buffer may be required. into which the “previous/correct” data is entered. And if the task ends with a breakdown and an error, then a “rollback” is made to these values.

kaiwas, 2024-05-12:
a simple idea emerged. Which will not be difficult to implement.
When a small, barely noticeable icon with an exclamation mark lights up in the view tree, the program takes a step back (Undo / AutoCtrl+Z) and displays a message that **"the “OBJECT NAME object” is broken. Repeat the operation with other values"**.

luzpaz, 2025-04-20:
This ticket could use a less generic title to help it being easily identifiable