FreeCAD/FreeCAD #13943: Sketcher Symmetry tool duplicates constraint labels

krushia, 2024-05-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After box selection of objects with a named constraint and doing Sketch > Sketcher tools > Symmetry, the named constrain is copied with identical name. This is normally not allowed. There will be an error `<class 'ValueError'>: Duplicate constraint not allowed` if the user enters an already used name in the constraint dialog.
[image]

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37100 (Git) AppImage
Build type: Release
Branch: main
Hash: ede6c811fd39d68a44949601f2eb7ecc8a67fdb7
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36
  * A2plus 0.4.64a
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * fasteners 0.5.20
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * workfeature
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.5.3
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-11:
@PaddleStroke FYI

PaddleStroke, 2024-10-23:
Symmetry tool also prints warnings

> 21:36:36  <Sketch> SketchObject.cpp(10045): duplicate geometry id 1 -> 5
> 21:36:36  <Sketch> SketchObject.cpp(10045): duplicate geometry id 2 -> 6
> 21:36:36  <Sketch> SketchObject.cpp(10045): duplicate geometry id 3 -> 7
> 21:36:36  <Sketch> SketchObject.cpp(10045): duplicate geometry id 4 -> 8

Other transform tools works correctly and don't have the described issue so that's already good. Regarding this symmetry tool I have actually not written the code that generate the geos so I'll need to discover this to be able to fix it

closed 2024-10-25 by chennes