FreeCAD/FreeCAD #13945: Part Design UI - NewSketch Plane attachment/orientation selection widget issue

marcuspollio, 2024-05-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a new Sketch in Part Design, the user has to choose a plane attachment/orientation either in the task panel or using the "widget" in the 3D view. While it works ok, the widget in the 3D view has an inconsistent behaviour: when doing a first sketch in a Body, the size of the widget is set depending on the zoom level of the 3D view, but adding further sketches display a different size with the former and between planes of different orientations.

To reproduce:
- Start FreeCAD and in Start choose Parametric Part (Part Design).
- In a Body, create a NewSketch. See the planes in the 3D view (first screenshot below).
- Finish your sketch and add features if you want to.
- Select the same body or the previous sketch and add a second NewSketch. See the size of the planes and how the YZ one is not matching the others in the 3D view now (second screenshot).

 
[image]

[image]

### Full version info

```shell
OS: GNU/Linux Ubuntu 22.04 LTS based (DE:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37243 (Git)
Build type: Release
Branch: main
Hash: 0095e52466c79c2135fce860a341554f3f51156c
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.5.1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-11:
I guess this issue is a combination from:
- https://github.com/FreeCAD/FreeCAD/issues/11133
- https://github.com/FreeCAD/FreeCAD/issues/12106
- https://github.com/FreeCAD/FreeCAD/issues/12584

I'll close this, please let me know otherwise.

closed 2024-05-11 by maxwxyz

marcuspollio, 2024-05-12:
@maxwxyz No worries =D
This issue was more about the orientation planes size differences tough, but if it's reworked, it will probably be solved.