FreeCAD/FreeCAD #13946: toolbar inaccessible outside screen

tcurdt, 2024-05-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

By mistake I have dragged the workbench toolbar. Now I can no longer attach it.
[image]

The console shows

`(qt.qpa.window) Window position QRect(1064,-280 70x1490) outside any known screen, using primary screen`

I am hope it's OK to report this here since it happened in Ondsel.

### Full version info

```shell
OS: macOS 14.4.1
Word size of Ondsel: 64-bit
Version: 2024.2.0.37191 (Git)
Build type: Release
Branch: (HEAD detached at 2024.2.0)
Hash: 2ad5fb327d068ae6280cf702d528b3a6e35770a7
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * Ondsel-Lens 2024.1.22.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-11:
It seems you are not using FreeCAD but the modified Ondsel version, is this reproducible in FreeCAD too?

tcurdt, 2024-05-11:
Good question. But I don't want to also break my FreeCAD install so I am not eager to try to reproduce this unless there is a way to revert this. Which I don't see yet.

furgo16, 2024-05-11:
If this is Ondsel-specific, another option might be to submit the issue at https://github.com/Ondsel-Development/FreeCAD/issues.

As an aside, I wonder if there is a way to transfer issues across GitHub projects. It might be useful to move issues around between upstream and distributions, both ways.

tcurdt, 2024-05-11:
> If this is Ondsel-specific, another option might be to submit the issue at https://github.com/Ondsel-Development/FreeCAD/issues.
> 
> As an aside, I wonder if there is a way to transfer issues across GitHub projects. It might be beneficial to move issues around, in both directions across upstream and distributions.

Seems like there is a way. Just not sure it works also across orgs.

https://docs.github.com/en/issues/tracking-your-work-with-issues/transferring-an-issue-to-another-repository

Feel free to move it if that's a better fit.

maxwxyz, 2024-05-11:
I'll close this here then, please create it at the Ondsel repo.

closed 2024-05-11 by maxwxyz

tcurdt, 2024-05-11:
FWIW ... Changing Settings > Workbenches > Workbench selector type/position brought this back into a usable state. Still a pretty ugly bug.