FreeCAD/FreeCAD #13947: [Sketcher] Automatic constraint for polyline on external plane reference always results in symmetric constraint

NomAnor, 2024-05-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I want to start a polyline on an external reference of a plane the automatic constraint will always create a symmetric (center) constraint which will snap the point to the origin of the plane. The normal line toll will create a "Constraint on object" constraint, which is what I would expect from the polyline tool aswell.

I only tried this in PartDesign, but I assume it the same for standalone sketches.

[image]

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36534 +178 (Git)
Build type: Debug
Branch: integration
Hash: b87e0ddfb5858c867abaecca3b3134ef2ababfe7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.3.0
Locale: German/Germany (de_DE)
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.60.0
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.0.0
  * Assembly3 0.11.3
  * QuickMeasure 2022.10.28
  * fasteners 0.4.21
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-11:
This is a known issue, previously it just snapped to the center and now with auto constraints it is creating a midpoint constraint

maxwxyz, 2024-05-11:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/11824

closed 2024-05-11 by maxwxyz