FreeCAD/FreeCAD #13954: Sketcher: arc length higher than half diameter looks bad

pierreporte, 2024-05-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In a sketch, draw an arc then constrain its length making sure it is longer than half the perimeter of a circle having the same radius. It looks like this:

[image]

While the result is rather clear when you think about it, it’s not optimal, if only because the longer the arc, the shorter the dimension look. It could be improved by making the dimension go around the arc, like angle constraints, while keeping the two lines parallel. Here is a proposal:

[image]

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37249 (Git) AppImage
Build type: Release
Branch: main
Hash: 3acbc561f299b4cddea8215c387a52f3d5132cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * BillOfMaterials-WB 0.0.7
  * CosmeticThread3D
  * OpenTheme 2024.4.20
  * Curves 0.6.35
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-05-11:
@FlachyJoe @PaddleStroke

FlachyJoe, 2024-05-12:
@maxwxyz you can assign me.

FlachyJoe, 2024-10-26:
@FreeCAD/design-working-group can you define the requested dimension display?

pierreporte, 2024-10-26:
@FlachyJoe the proposal is the right display.

PR #17720 (2024-12-09): Fix Arc length look for angle > pi
Fix  #13954

@pierreporte FYI

[image]
closed 2024-12-09 by chennes