FreeCAD/FreeCAD #13955: Materials: Source for Mechanical properties of Wood

ShamanTcler, 2024-05-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Finding "validated" material properties can be hard. This link: https://www.fs.usda.gov/research/treesearch/62244 , from the US Dept of Agriculture, Forest Service seems to have sufficient credibility. 

The information is available as a PDF: https://www.fpl.fs.usda.gov/documnts/fplgtr/fplgtr282/chapter_05_fpl_gtr282.pdf

With some help, I am willing to assist with the material creation.

### Full version info

```shell
I am currently building the dev version on windows
```

### Subproject(s) affected?

None

### Anything else?

[chapter_05_fpl_gtr282.zip](https://github.com/FreeCAD/FreeCAD/files/15283126/chapter_05_fpl_gtr282.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-05-11:
I'll just add that wood has orthotropic mechanical properties and in most cases should be modeled with such an approach. Unfortunately, orthotopic materials are not yet supported in FreeCAD FEM but the new Material system laid the foundations for further development in this regard. So it might be better to implement orthotopic linear elasticity first and then add more wood materials to the library.

pierreporte, 2024-05-11:
@davesrocketshop what do you think about this?

ShamanTcler, 2024-05-11:
I am assuming that material props are going to be useful everywhere. i.e. given density, what does this assembly weigh?

I certainly could see locking out wood as a material for FEM.  Though I was quite surprised to see the detail of the mech properties supplied by the forest service....

I will also add I was pleasantly surprised by FreeCad's FEM capability.  I took a class at a Junior College in Machining (ran 3 & 5 Axis Haas machines). As part of my "project" I did a FEA model of my part and included it in the presentation.  The instructor was very pleased chuckle.

Yes retired Mech Es (wrote software for other engineers to use) still take classes and learn ;)

davesrocketshop, 2024-05-11:
> @davesrocketshop what do you think about this?

Wood is a notoriously horrible product from a mechanical point of view. There is so much variability that published results will rarely match your wood, and it can change based on humidity and so many other variables. That being said, these do look like reasonable defaults.

Double check the license. US gov sources are generally good, but we don't want to include restricted data.

ShamanTcler, 2024-05-11:
I emailed sm.fs.rdpublicweb@usda.gov ..... owners of the website.  They will hopefully respond.

ShamanTcler, 2024-05-21:
Got a response from Witne Neil today.
[image]

davesrocketshop, 2024-05-21:
> Got a response from Witne Neil today. ![WitneNeilEmail](https://private-user-images.githubusercontent.com/30176093/332566027-1c688add-d0ad-4455-9707-0e7429f7c834.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYzMjIxNjcsIm5iZiI6MTcxNjMyMTg2NywicGF0aCI6Ii8zMDE3NjA5My8zMzI1NjYwMjctMWM2ODhhZGQtZDBhZC00NDU1LTk3MDctMGU3NDI5ZjdjODM0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIxVDIwMDQyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZkNmI3NzdiZWM0ZjM2MTFlN2I3OWZhYzY2NTA2NjRhNTY4MzgzMzJlMWRjODZkNjYyNmJhMmJkMTMzZGQ5Y2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.60EsRMmbps9XG4iEGYYljI_lqAJ0mPIPLeYaIV5pBmY)
> 
> Can I assume this is sufficient?

Absolutely. Just cite him as requested.

luzpaz, 2024-08-07:
@ShamanTcler any progress ?

ShamanTcler, 2024-08-07:
Not on my end, A while ago I looked for doc on the materials work and did not find much. I will be going to the meetup in Springfield next week, hopefully some progress there.

davesrocketshop, 2024-08-07:
This may be more than you need, but it gives a deep background for developers

https://youtu.be/PNetPYZlf9o

ShamanTcler, 2024-08-07:
Very nice thanks ... I watched ~5 min but will finish later

FEA-eng, 2024-08-07:
Related forum discussion: https://forum.freecad.org/viewtopic.php?t=89564

ShamanTcler, 2025-06-25:
I started to look into actually importing some data but ran into a few issues.

1)  Was not sure of to do a bulk import of data, and what the "PR" process would look like

2)  The whole idea of display/rendering came into play ... choosing Cherry should show grain not just a uniform reddish color

3) Being a retired Mech Engineer who has not played the stress/strain game in a while the different terminology scares me enough to want a second set of eyes .... what's the phrase "use it or lose it"

gholmberg, 2025-06-26:
I made this [Google spreadsheet](https://docs.google.com/spreadsheets/d/13yIXwL_B-HZFSiqNq6ucwPPLw5BYY4gEkpEvLdBN-Rg/edit?usp=sharing) of about 250 wood species.  wood-database.com has more than than, but I don't know how many.  I can export this as CSV.

Greg

davesrocketshop, 2025-06-26:
Give me a few days to see what's possible. I can definitely work with this.

gholmberg, 2025-06-26:
Wow, that's amazing!  Really impressive work.  Thank you so much.

You might have guessed, these 250 are species used for building guitars.  There are 350 more species on wood-database.com  These are averages for each species.  Wood is a highly variable material, varying +- 50% between trees of the same species, and even within a tree.  Averages are still useful though.

As you observed, the 'onshape' sheet is a subset of the data in the 'All' sheet: all the species, but only a few columns. From cell B2: =query(AllSpecies, "select A, L, O, M*1000000000, Q*1000000, 0, R*1000000, 1") These are just the properties that the Onshape CAD software would accept.

But columns A through V are all useful.  The columns after V are calculations I did, so probably don't belong in a materials database.

Here are some properties I think might be generally useful.  Certainly useful for me to calculate the rigidity and frequency of vibration of acoustic guitars.

[Wood Model](https://docs.google.com/spreadsheets/d/1Jp6ASOjuEbyzGkG3HO61zWH_UDZ4Lf6eewI11v7gOmA/edit?usp=sharing)

As I'm sure you know, wood is orthotropic, so requires multiple values for Young's modulus (3), Poisson's ratio (6), Compressive strength (2), Shear modulus (2), shrink (4).

We normally think of wood as having three directions: longitudinal (along the grain, vertical in the tree), radial, and tangential.

wood-database only has values for longitudinal, but the USDA Wood Handbook has all these values for certain common species in North America.  And I use several of these orthotropic properties in my calculations. Unfortunately, I know of no published values for many species, and so I've used averages for those (orange in the spreadsheet, columns labeled 𝜈LR and 𝜈RL).

Given the variation within a species, I will be measuring individual pieces of wood in my shop to help me determine thickness for the acoustic guitar top (there's a formula for that that uses E_long, E_cross, and G shear, derived from frequencies I measure), and to determine guitar top stiffness/frequency.  For this, my plan in FC is to create new instances in the Materials database under 'User' for these individual pieces of wood, and then reference those in my 3D model, and use them in the FEM workbench to see the modes and frequencies of vibration.

So these one-off instances would inherit from the System materials, I guess?  Not sure

Well, anyway, I thought it might be useful to understand my use-case.

Greg

gholmberg, 2025-06-29:
OK, I think I've taken the [Wood Model](https://docs.google.com/spreadsheets/d/1Jp6ASOjuEbyzGkG3HO61zWH_UDZ4Lf6eewI11v7gOmA/edit?usp=sharing) as far as I can. I studied the system standard .yml files, and tried to make the wood model conform as best I could.  I'm proposing two new models (Wood and Shrink), and changes to two existing models (Orthotropic Linear Elastic and Costs).  I've included two generic materials with average values (Hardwood and Softwood), and three example species (Sugar Maple, Sitka Spruce, Douglas-fir).

But I'm new to FC, so I think someone who knows what they're doing should review it.

Now I'm working on transforming my data in the spreadsheet into something that conforms to my proposed model. Here it is with the three [example species](https://docs.google.com/spreadsheets/d/1KGd7NuLZteOg5eSeiPs7HPhQ6kc69PjGjl6H2VJ3aF0/edit?usp=sharing).

The way I envision this working is that the first three rows indicate the property.

The first row would be the model folder, if that's the right word.  For example: General, Models/Costs, Models/Mechanical, Appearance, etc.

The second row would be the the model name.  For example, Costs, Orthotropic Linear Elastic, Texture Rendering, etc.

The third row would be the specific property name.  For example, Author, Vendor, YoungsModulus, AmbientColor, etc.

Rows below that are the actual data values.

I think this could be a general bulk data import tool. You could put any property names across the top that you want, and a Python script could convert it to Materials YAML files without any additional information, simply by reading the first three rows.  Any kind of material, using any set of models.

Thoughts?

Greg

davesrocketshop, 2025-06-30:
Some comments:

For data import, the spreadsheets should include raw data only as opposed to calculations as much as possible. There are a few reasons for this.

Let's take your example of mass. This is not a calculation that should be done in the material card. That's something that should be done in FC with the density as input and the volume calculated from the object. So density should be specified but not mass.

Material cards are considered immutable. For example, changing the density value would change mass values for existing projects. This could have unpredictable results in your FC projects. I do my best to prevent these changes in the editor. You can create a new material with the changed value, but the original should never change. What can change? Non-property values such as author, description, etc. You can _add_ new property values, but not change existing ones. So if a property is unknown, it is better left blank so that it can be added in the event it becomes known. Materials with _changed_ properties are considered new materials, so should have a different name and UUID.

On a more practical level, the python libraries return the formula when retrieving the value of a cell, not the evaluated result. As far as I've been able to tell it's left as an exercise for the alert reader on how to do that evaluation. It's easy in the spreadsheet, but not so easy for the importer. In the case of your original spreadsheet there were only 4 calculated values used for the materials, so that is manageable.

So what does this mean from the POV of your spreadsheet? Given that for a lot of your materials you're using averaged or typical values rather than actual values, I suggest that this should produce two materials. The first leaves unknown values blank, allowing them to be filled in later as an authoritative reference becomes available. The second would include the suggested or inferred values based on your calculations. In this way, they're differentiated allowing the user to have clear knowledge of how the values were obtained, and allowing the materials to be refined in future. Or at the very least include something in the model name to inform them that the values aren't perfect. For example, something like "White Fir" and "White Fir (Averaged Values)". They're still useful and should be included, but we should be a little pedantic if we're creating a curated data set. I'm not referring to cases like bamboo with widely varying densities. An average is appropriate there. I'm referring more to properties such as Poisson ratios, were values are based on properties of other woods.

Including costs is problematic for many reasons. They change with time for example. And your costs will be very different than mine due to country, shipping, etc. It's good to include the model, but the values should be left to the user.

Like materials, models are immutable for similar reasons. Creating new models is the answer in some cases but not all. Materials can leave entries blank. The FC code has routines to select materials that have values for specified properties, so it can handle this easily. Wood and Shrinkage are good model candidates. The existing linear elastic model is good as is and can't be changed anyways, but there should probably be a new model with the anisotropic properties that are lacking. Properties that don't apply are left blank.

I like your idea of a bulk import tool, but that may be overkill here. The wood database is targeting a single spreadsheet which means both the spreadsheet and the importer can be very simple. Let's leave the generic tool as an exercise for another day.

I've also been looking at a curated spreadsheet based on your original one (https://github.com/davesrocketshop/Woods/blob/master/Resources/Data/Wood%20Properties%20V2.xlsx). Changes are minimal compared to your bulk importer approach.

It removed a few things and added some others to keep things simple. As mentioned earlier, I've removed cost information, as well as many of the guitar specific pages of interest to you but not interesting from a material perspective. I've also made the rows more consistent.

I've also added images for each file taken from wood-database.com. We don't support textures in FC (yet) but they are supported in materials. Even if the images can't yet be displayed in FC, I can use them to get an averaged color value to set the appearance. In the meantime I've sent in a request to the site admins for permission to use. If they say no (or don't respond) we can include the averaged color values only.

Alternate names, including latin names, are added as tags. This is another currently unsupported FC feature that will be added in future. This will (someday) allow the user to search using the alternate names.

Names have been made more consistent, for example "Redoak" becomes "Red Oak", whereas "Rosewood" remains unchanged. The primary name is the same as in wood-database.com. That was true for the bulk of the woods, but there were a small number of woods that were different. When generating the materials, they are converted to title case for consistency, for example "White fir" becomes "White Fir".

I've added columns for the generated UUIDs. This will allow the importer to be rerun without changing every material. The importer automatically detects generated values and will create two materials as mentioned earlier. The importer itself is still a work in progress but should be stable by the end of the week.

I think there were 3 woods that weren't in wood-database.com. I elected to exclude them as there were no images available, and for consistency, but I could easily be convinced to re-add them. They definitely fall into the obscure category. They might be good candidates for user managed materials but again I can be swayed.

I'll add a curators guide to the repository that will give guidelines on how to add new wood entries.

davesrocketshop, 2025-07-05:
I think we're closer to each other than we realize. I would love to have a video call to discuss if you're available. I've stuff on the go most of the weekend but if you're free next week I can arrange something. I'll be generally available Monday, Thursday, and Friday, with evenings available Tuesday and Wednesday. I'm EDT here.

The sad thing about the Python library is "values_only" returns the formula, not the value. Le sigh. A lot of the calcs can be done in the importer though so not really a problem.

I'm liking most of your model ideas. Wood though is technically anisotropic, not orthotropic. Ortho uses a Cartesian coordinate system. Creating an appropriate model is trivial... it just means we can't use the one that is there. That's in no way a show stopper. Some of the others may make sense to break down further. For example, instead of having one big model to describe everything, shrinkage may be better as a separate model.

Averages are ok in the context of wood's inherent variability. I'm more concerned about using an averaged value from other types of wood for an unknown parameter.

Machining is an interesting case. I think the "standard" values are meant to be starting points. I think it, and costs, make the case for a new type of "variant" or "volatile" property that can change without affecting other projects, other than creating up to date cost breakdowns or tuning of machinability parameters. FC doesn't support that currently, but I know a guy :)

Similarly 2D arrays don't currently support strings. It's on the desired features list but is low priority at the moment.

There are a lot of ways you can go with all of this. This is why I feel a video discussion might be the way to go.

gholmberg, 2025-07-05:
That sounds good.  I'm retired, so I'm available any time after, say, 10 AM PST (California).  I have no plans for next week.

Greg

luzpaz, 2025-07-06:
CC @mnesarco i bet you'd appreciate this discussion (author of the https://github.com/mnesarco/MarzWorkbench workbench)

gholmberg, 2025-08-01:
Dave--

OK, sorry for the delay (family problems).  I took your Excel spreadsheet and added some additional columns (mostly from the USDA), then filled in three example species (Hard maple, Sitka Spruce, and Douglas fir).  Here's a Google [sheet](https://docs.google.com/spreadsheets/d/1Cp3FobgHrRwZWSvj_Q0kMC9BSWwCpPUyFrQdfcEg0xQ/edit?usp=sharing).  When you go there, you can request edit permissions.

The column order is a bit weird, but I know your code depends on what was already there, so I only appended new columns.  Feel free to re-order them though.

Greg

davesrocketshop, 2025-08-19:
My turn for the delay. I've been enjoying the summer weather :)

I'm back on this now and should have something to show in the coming days. I haven't re-ordered the columns, but I would suggest you consider an order that makes sense to you. I have the columns defined as constants so changing the order in my code is not that big of a deal.

davesrocketshop, 2025-08-20:
First pass is complete. I'll probably need to tweak the code as different woods are added. Error handling is minimal so there may need to be added robustness as special cases are encountered. I did add another column to handle the species name, and moved the name from the additional names column. This allows me to complete the botanical fields.

I've created a number of models to handle specific features (botanical, sound, shrinkage, etc.) You can see the models and materials here: https://github.com/davesrocketshop/Woods

You can also check this out using git if you're familiar with that process and use try it in FreeCAD. I won't be publishing it for the addon manager until we've got most of the woods entered and are a little more confident with the results. Or you can manually copy the models and materials to your user folder. Let me know if you need help with that.

gholmberg, 2025-08-20:
That looks good, Dave.  Thank you for doing that.

Small nit: in Sound.yml, the units should be 'm^4/kg*s' not 'm^4/kg/s'

I'll start work on getting all the data massaged into the spreadsheet.

davesrocketshop, 2025-08-20:
m^4/(kg*s) == (m^4/kg)/s == m^4/kg/s in FreeCAD unit descriptions. Unless you mean (m^4*s)/kg

gholmberg, 2025-08-27:
OK, it's ready.

davesrocketshop, 2025-08-30:
Updated materials ready for inspection

gholmberg, 2025-08-30:
OK, now, how do I use these materials?

I can assign a wood material to a part.  It will display using the diffuse color--good.

But how can I use the texture image?  Only in the render workbench?

Now, how do I get the numbers from a material?  I have two scenarios.

1. In an FC Spreadsheet, I want to get a part's material, and then get that material's Density (and write some formulas).  Let's say the part is from a Pocket operation, Label=Pocket001, Label2=Sides.  I tried this in the spreadsheet:

=<<Pocket001>>.ShapeMaterial

But I was unable to get any further properties.

2. In Python I want to do calculations using a part's material's Density.  I will also read other values from a spreadsheet, and write values back to the spreadsheet.

So I need API docs for Material objects and Spreadsheet objects.  Maybe I'm just particularly bad at googling, but I couldn't find these API docs.

Thanks,

Greg

davesrocketshop, 2025-08-30:
Updated. I've incorporated the LinearElastic into Wood to reduce duplication.

Texture images aren't currently implemented. It's high on the development priorities but will not be in 1.1

To get the properties, such as density, do this:

```
obj.ShapeMaterial.getPhysicalValue("Density")
obj.ShapeMaterial.getPhysicalValue("Density").UserString for the string representation
obj.ShapeMaterial.getPhysicalValue("Density").Value for the float value in internal units

```

See the unit tests for a lot of examples: https://github.com/davesrocketshop/FreeCAD/blob/main/src/Mod/Material/materialtests/TestMaterials.py

I've never used the spreadsheet in Python so I'm not much help there. There are some tests in the src folder. I'd start there: https://github.com/FreeCAD/FreeCAD/tree/main/src/Mod/Spreadsheet

gholmberg, 2025-08-30:
> > Moving on to the FCMat files themselves...
> > ```
> > 1. [Bug] It seems like changes I made to capitalization of altnames was lost?
> > ```
> 
> Yes. That was done for consistency. There is still inconsistent capitalization being used. Is this important?

Well, it has no functional consequences, but I thought we should be consistent.  I was going for consistent initial caps for each word in the Name and the Tags.  I guess I missed some?

Species names only init-cap the first word.  It seems to be some sort of standard in the biology world.  

> > ```
> > 2. [Bug?] The five ratios (ET/EL, ER/EL, GLR/EL, GLT/EL, GRT/EL) are no longer in the file.  But the multiplied-out values look good.
> > ```
> 
> Are the ratios still useful? I would think the multiplied values would be what is actually used.

I think the ratios are useful.  The USDA probably did it this way due to the huge variability of samples within a species, so one can measure just EL of a sample and estimate the other five (apparently there's a strong correlation). So if we included a value for "Shear Modulus Long->Radial / Flex Modulus, Long" (unit-less), then people can use this number with their samples. 

For example, in the acoustic guitar workbench I plan to write, the user would measure (deflection or frequency) the EL of his sample, and the workbench would use these ratios to estimate ET, ER, and GLT, which I use to calculate the resonant frequency of the guitar's top, and the frequency response of the completed guitar.

Of course, I could reverse these ratios from the numbers that are already there, so you could convince me to do that instead.

gholmberg, 2025-08-31:
> Updated. I've incorporated the LinearElastic into Wood to reduce duplication.

It looks good to me.  Let's release it.

davesrocketshop, 2025-08-31:
> Well, it has no functional consequences, but I thought we should be consistent. I was going for consistent initial caps for each word in the Name and the Tags. I guess I missed some?
>
> Species names [...]

I can do init caps easily. I already do similar elsewhere. I'll make that change.

> > > ```
> > > 2. [Bug?] The five ratios (ET/EL, ER/EL, GLR/EL, GLT/EL, GRT/EL) are no longer in the file.  But the multiplied-out values look good.
> > > ```
> > 
> > 
> > Are the ratios still useful? I would think the multiplied values would be what is actually used.
> 
> I think the ratios are useful. The USDA probably did it this way due to the huge variability of samples within a species, so one can measure just EL of a sample and estimate the other five (apparently there's a strong correlation). So if we included a value for "Shear Modulus Long->Radial / Flex Modulus, Long" (unit-less), then people can use this number with their samples.
> 
> For example, in the acoustic guitar workbench I plan to write, the user would measure (via deflection or frequency) the EL of his sample, and the workbench would use these ratios to estimate ET, ER, and GLT, which I use to calculate the resonant frequency of the guitar's top, and the frequency response of the completed guitar.
> 
> Of course, I could reverse these ratios from the numbers that are already there, so you could convince me to do that instead.

Well, realistically we should do one or the other, not both since the other can be calculated easily. Which one I'll leave up to you.

davesrocketshop, 2025-08-31:
Tags are now capitalized.

And the pull request to make it an official add on has been submitted. Yay!

gholmberg, 2025-08-31:
Regarding my attempts to access material data from a spreadsheet, I've discovered that FC expressions support this:

`=<<Pocket002>>.ShapeMaterial.Name`

produces "American Beech".  Yay!

`=<<Pocket002>>.ShapeMaterial.PhysicalProperties`

shows this result in the spreadsheet:

[image]

So it's a Python dictionary.

Unfortunately, it appears that FC expressions do not support dictionaries.  So you can't write this:

`=<<Pocket002>>.ShapeMaterial.PhysicalProperties[Density]`
or this
`=<<Pocket002>>.ShapeMaterial.PhysicalProperties['Density']`

There's a ticket [getting Material properties dictionary values in spreadsheet cells](https://forum.freecad.org/viewtopic.php?t=16714&start=70) and a [github issue](https://github.com/FreeCAD/FreeCAD/issues/5771), but it appears unresolved, I think.  I can't quite figure out if realthunder did something about dictionaries in expressions, and then forgot about it, or what.

So close!  Arrgh.

Greg

davesrocketshop, 2025-08-31:
Yes, it's a dictionary similar to what was in the previous materials version and is there for compatibility. But the dictionary doesn't support newer non-trivial properties such as arrays.

I'll confess that I hadn't thought about working with expressions. The github issue was actually linked from our previous bug tracker, appearing to be from 2018 or earlier. It's also been closed as not intended. You might want to create a new issue.

gholmberg, 2025-08-31:
My confusion is that realthunder said this [pull request](https://github.com/FreeCAD/FreeCAD/pull/3062) fixes it.

OMG, I just got it to work!

`=<<Pocket002>>.ShapeMaterial.PhysicalProperties[<<Density>>]`

shows "7.21e-07 kg/mm^3"

Unfortunately, I think it's just a string.  I can't use it in a numerical expression.  :-(

Maybe I can find an expression operator to turn a string into a Quantity...

davesrocketshop, 2025-08-31:
Try PropertyObjects instead of PhysicalProperties. Entries have a Value property that will contain a Quantity (or Float, or whatever)

gholmberg, 2025-08-31:
Yes!  This works.

`=<<Pocket002>>.ShapeMaterial.PropertyObjects[<<Density>>].Value`

displays `721.00 kg/m^3`

If I have this in cell A1, then in cell A2 I can have: `=A1 / 2`

which displays: `360.50 kg/m^3`

Woohoo!  I can write formulas in spreadsheets using Materials data.

Greg

gholmberg, 2025-08-31:
I'm just realizing the cause of this:

[Bug] I also noticed that properties that the model defines as Float (SteamBendable) display as '0', even though there's no entry for it in the FCMat file. 

It's because both Hardwood and Softwood have a value defined (67 or 0), so if a material doesn't have it defined, it gets one.  It's never empty.

We should undefine SteamBendable in Hardwood.FCMat and Softwood.FCMat.  There's no reasonable default.

davesrocketshop, 2025-09-01:
Done, but only because the add on is new. Values can be modified but should never be removed as it can break projects using the material.

gholmberg, 2025-09-09:
Our new wood materials don't work the way I expected.  I assigned a wood material to a Part, and then asked the Part for its mass: `<<MyPart>>.Shape.Mass` The number I got was calculated using a density of 1000 kg/m^3 (i.e. like water), and not the density of the wood (370 kg/m^3).  So the Part has ignored the density of the material assigned to it.

I found this feature request, open since 2024, which I think covers this issue: 

[Measurement: Measure volume + mass with material of the object](https://github.com/FreeCAD/FreeCAD/issues/13715)

As a work-around, I calculate mass like this:

`=<<MyPart>>.Shape.Volume * 1 (mm ^ 3) * <<MyPart>>.ShapeMaterial.PropertyObjects[<<Density>>].Value`

Greg