FreeCAD/FreeCAD #13956: improvement of the "hide" function

4a3, 2024-05-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

it is inconvenient to work with objects when there are many of them. it is better to be able to hide unnecessary things with a "short cut"

### Full version info

```shell
please supplement the "hide selected object" function ("space" button) with the "hide everything except the selected one" function ("ctrl + space" button) and the "show all" function ("alt + space" button)..
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-05-11:
Do you mean something similar to what is presented in #12142?

maxwxyz, 2024-05-11:
duplicate of https://github.com/FreeCAD/FreeCAD/issues/12142

closed 2024-05-11 by maxwxyz