FreeCAD/FreeCAD #13958: Assembly: wrong mating of cones

pierreporte, 2024-05-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Mating conical parts is very common, especially for screws:

[image]

The assembly workbench can work with cones but their surfaces are not coincident. The file attached contains one assembly file and two parts. The parts are designed so that the male cone is centered around the female part. As you can see in the assembly file and the picture below, there is an offset when using a distance joint or a fixed joint. It is necessary to apply a distance joint with the correct value to two parallel surfaces to get the right result, but it is not necessarily possible because the distance is usually unknown and there may be no available surfaces in the first place.

[cone mating.zip](https://github.com/FreeCAD/FreeCAD/files/15283487/cone.mating.zip)

[image]

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37249 (Git) AppImage
Build type: Release
Branch: main
Hash: 3acbc561f299b4cddea8215c387a52f3d5132cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * BillOfMaterials-WB 0.0.7
  * CosmeticThread3D
  * OpenTheme 2024.4.20
  * Curves 0.6.35
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Assembly

### Anything else?

This issue was raised in #12445, which was only partially completed.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-05-13:
To achieve what you want, it needs a distance joint of 0 between the 2 conic surfaces.

And currently distance joint doesn't work between 2 conic surfaces as the solver currently doesn't have this.

Another solution would be to make it possible to select the cones 'top' points. ie when selecting the conic surface, the JCS would position itself at the 'top' point. Then you would be able to match 2 cones by making a fixed joint between the 'tops'. This I think I should be able to do.

Now after thinking about it, the distance joint between 2 conic surfaces, is basically a distance between the tops points. Maybe I can cook something.

pierreporte, 2024-05-13:
If you know the distance between the point ($\ell$) and the angle ($\alpha$, equal for both cones), you can calculate the distance between the two surfaces ($h$) using this simple relation: $$h=\ell\times\sin\alpha$$

The distance $h$ is closer to the intent of the user.

If the angle is different for both cones, it’s harder. Let $\alpha_m$ and $\alpha_f$ the angles of the male and female cones respectively.

If $\alpha_m<\alpha_f$, you need to determine the minimum diameter of the conical hole on the female geometry, and determine the position of this diameter on both cones, then set the distance $\ell$ so that the intersection of the two cones is a circle of the same diameter. If $\alpha_m>\alpha_f$, then you need to determine the maximum diameter of the conical shape of the male geometry, and adjust the distance cones the same way. You will then have contact with the two parts.

If you want clearance the same way as defined for $\alpha_m=\alpha_f=\alpha$, more calculations are needed to determine $h$.

PaddleStroke, 2024-05-14:
Ah yes its simple only if both cones have the same angle. And only to make distance between them when they are aligned.
But cone distance is actually much harder than this indeed. The cones don't even have to be aligned.
[image]

So what I can do for now, that would solve the case you describe, is that when you select a cone, the JCS position itself to the top point.

pierreporte, 2024-05-14:
Seeing your picture, I think that there should be a distinction between mating joints and tangent joints. Both can have a distance. They can sometimes be equivalent. Mating is for moving both surfaces to make them overlapping, while tangent is for making them touching each other.

commit 7c2a95bb (2024-05-20): Assembly: JCS selection: cones surfaces offers the cone's apex. Fixes #13958
closed 2024-05-20 by chennes