FreeCAD/FreeCAD #13961: [Problem] Pocket with "Start At Center" still starting from Edge

tcurdt, 2024-05-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the CAM/Path workbench I want to cut a pocket. Despite setting "Start At" to "Center" as [suggested in the documentation](https://wiki.freecad.org/CAM_Pocket_Shape) the pocket is still cut from the boundary - and **not** the center. It seems the setting just gets ignored.
[image]
[image]

### Full version info

```shell
[code]
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * ExtremeProDark 2.6.4
  * ExtremeProLight 1.4.3
[/code]
```

### Subproject(s) affected?

CAM/Path

### Anything else?

If necessary I am happy to provide the project file.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tcurdt, 2024-05-11:
Same problem with Pocket3D operations.

tcurdt, 2024-05-11:
Funny enough, when I switch how multiple features should handled to "individually" it works for Pocket3D operations.
[image]
[image]

maxwxyz, 2024-05-12:
Could you try this in the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

tcurdt, 2024-05-12:
> Could you try this in the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

It's repeatable for me in the weekly build as well.

luzpaz, 2024-08-12:
We need full [About](https://wiki.freecad.org/About) info please for the dev version, TIA!

luzpaz, 2024-08-12:
@Russ4262 do you have a moment to weigh-in and confirm ?

tcurdt, 2024-08-12:
@luzpaz again confirmed with

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38429 (Git)
Build type: Release
Branch: main
Hash: 362cf1c80edb89957d81c0dbe6e50ce35daaa6db
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * ExtremeProDark 2.6.4
  * ExtremeProLight 1.4.3
```
[image]

Russ4262, 2024-08-13:
@tcurdt ,
Evening.  Are you able to provide a source file, or simplified example.  I am testing on Windows with simple rectangular shapes and 'Offset' cut pattern, and I see a successful switch from starting at Edge to starting at Center.  I will have to test with more complex shapes.
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38429 (Git)
Build type: Release
Branch: main
Hash: 362cf1c80edb89957d81c0dbe6e50ce35daaa6db
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * FC_SU
  * freecad.gears 1.0.0
  * freecad.pathred
  * Z_MacroStartup
```

luzpaz, 2025-07-17:
@tarman3 thanks. Do you mind reviewing the documentation to see if it's in order ? https://wiki.freecad.org/CAM_Pocket_Shape

tarman3, 2025-07-17:
I do not want to say that it is not a bug
But this behavior in written in C++ code and I do not see any chance that this will be fixed in near future

tcurdt, 2025-07-17:
IIUC the idea was also to keep the cpp code as light as possible and do most of the things in python (probably because of the reason you mentioned).

`StartAt` and `MinTravel ` I have only found on the python side of things.
Can you elaborate a bit more what needs fixing?

I would offer to help - but I am still a little lost in the codebase.

tarman3, 2025-07-17:
https://github.com/FreeCAD/FreeCAD/blob/9cd3b31067b488a646046a36af47168331e5644b/src/Mod/CAM/Path/Op/Area.py#L312

All what happening inside this is miracle for me

tcurdt, 2025-07-17:
Maybe @sliptonic can guide us a little to find the right place to fix this.

closed 2026-09-28 by Connor9220