FreeCAD/FreeCAD #13969: Navigation cube cannot be moved on 0.21.2

fenrir-5, 2024-05-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi everyone,
After a quick search, I found few relevant topics on FreeCAD forum but the problem seem to be related to AMD graphics. I'm using Intel graphics and everything worked great few months ago when I last used FreeCAD.

### Full version info

```shell
[code]
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * LCInterlocking
  * A2plus 0.4.62
  * sheetmetal 0.4.0
  * ExplodedAssembly
  * Render 2024.1.23
  * ArchTextures
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-12:
Is the setting movable navigation cube checked?
Could you try if the issue persists in the current weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

fenrir-5, 2024-05-12:
That option seem to be missing in Preferences > Display > Navigation tab > Navigation cube section. In report view there are some error which I don't think they have something to do with the Navigation cube situation:
`
12:52:14  Path override failed for key base::DIR_APP_DICTIONARIES and path '/usr/lib/freecad/bin/qtwebengine_dictionaries'
12:52:14  The `.FCStd#` backup format is deprecated as of v0.21 and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
`

I'm a bit scared of trying weekly builds, or at least not until I finish the project I'm working on now.

liamhays, 2024-05-12:
This is probably the same issue as this: https://gitlab.archlinux.org/archlinux/packaging/packages/freecad/-/issues/3, which is a product of Arch's switching the FreeCAD build to Qt6.

maxwxyz, 2024-05-12:
could this be the same issue as https://github.com/FreeCAD/FreeCAD/issues/13494#issuecomment-2106262038

fenrir-5, 2024-05-12:
Yep, i'm sure that's the issue. Problem is I can't downgrading it without falling into the Python rabbit hole so I guess waiting for the next release is best I can do.

With the 0.21.2-6 I get:
`freecad: error while loading shared libraries: libpython3.11.so.1.0: cannot open shared object file: No such file or directory`

maxwxyz, 2024-05-12:
I'll close this then, you can comment in the other issue

maxwxyz, 2024-05-12:
Duplicate of #13494

closed 2024-05-12 by maxwxyz