FreeCAD/FreeCAD #13974: Make Marker Size freeform / spin box input

Oozlum, 2024-05-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Marker size preferences currently allow you to choose only from a fixed selection of sizes. On a high resolution display even the largest size is too small for my liking. I have manually modified my preferences file to set the marker size to 30px, which works and looks much better, so the existing restriction appears to be limited to the UI design only.

Can we please change this option to use a spin box or freeform entry instead?

### Full version info

```shell
V0.21 and v0.22 weekly builds. Issue posted on mobile and will update with full info later.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-12:
related: https://github.com/FreeCAD/FreeCAD/issues/13681

Syres916, 2024-05-12:
@Oozlum it maybe best if you look at how the last size increase was done to explain why a spinbox isn't applicable to the current solution, see https://github.com/FreeCAD/FreeCAD/commit/00826abe7a14c7988f0a317f658222be9bb9071d

Syres916, 2024-05-12:
> I think my question would be then, if that commit implemented 20, 25, and 30px sizes, why aren’t these available to select?

Being specific with my wording so as to reduce ambiguity, using:

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37249 (Git)
Build type: Release
Branch: main
Hash: 3acbc561f299b4cddea8215c387a52f3d5132cf6
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.4.13
  * Design456 0.0.1
  * Rocket 3.3.0
  * CfdOF 1.24.8
  * Assembly3 0.12.0
  * fasteners 0.5.20
  * freecad.gears 1.1.0
```

30px markers are easily to select with a Pick Radius of 20px but you are maybe struggling to select them with the default Pick Radius of 9px. Pick Radius is available through Edit > Preferences > General > Selection

Obviously 0.21.x doesn't have the new sizes available as they are classed as a feature enhancement and the general rule is only bug fixes are backported from the dev (main) branch to the stable branch.

Oozlum, 2024-05-12:
Thank you.  It isn't so much the selecting that was the issue for me, rather that with the smaller sizes the pre-selection wasn't as clear.  Either way, 30px is fine for me, so I'll close this given your comments on its implementation.

closed 2024-05-12 by Oozlum