FreeCAD/FreeCAD #13975: Improve pre-selection visibility

Oozlum, 2024-05-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On a high resolution display pre-selection of lines and vertices is very difficult to see. There doesn’t seem to be a single choice of pre-selection colour that works well for all objects.

Could we perhaps use a high contrast / reverse video effect for pre-selection rather than a single nominated colour? It would also be helpful if we could add some kind of bloom, or enlargement of preselected objects to make them stand out.

### Full version info

```shell
v0.21 and v0.22 weekly build. Will update with full info later.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-05-13:
I did think about this earlier this year but it was more about selection than pres-election. The idea was to add a border around the selection, without covering it, similar to the following image. For pre-selection, it could work the same but with the pre-selection color. It could also work in sketches. This selection style is used in Catia Composer.

[image]

Oozlum, 2024-05-13:
Yes, that might work quite well.  Frankly anything that makes the object bigger and differentiates it would be a help.