FreeCAD/FreeCAD #13977: Crash during Path->Face

Krandelbord, 2024-05-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Path Mode I selected face operation, after clicking apply it crashed

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c510) [0x7f6d4645a510]
#1  0x7f6d478a10d7 in QLabel::event(QEvent*) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Widgets.so.5+0x7
#2  /tmp/.mount_FreeCAdAUZOv/usr/lib/python3.10/site-packages/PySide2/QtWidgets.cpython-310-x86_64-linux-gnu.so(+0x4b5ba3) [0x7f6d1f8b5ba3]
#3  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libpython3.10.so.1.0(+0x148741) [0x7f6d48948741]
#4  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4ba8) [0x7f6d4893d758]
#5  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x75) [0x7f6d48949305]
#6  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x739) [0x7f6d489392e9]
#7  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x75) [0x7f6d48949305]
#8  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x739) [0x7f6d489392e9]
#9  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libpython3.10.so.1.0(+0x155332) [0x7f6d48955332]
#10  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libFreeCADBase.so(PP_Run_Method+0xe3) [0x7f6d47fd55d3]
#11  0x7f6d47f2692b in Base::InterpreterSingleton::runMethodVoid(_object*, char const*) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libFreeCADBase.so+0x2b
#12  0x7f6d491449b3 in Gui::PythonCommand::activated(int) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libFreeCADGui.so+0x53
#13  0x7f6d49145108 in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libFreeCADGui.so+0x2a8
#14  0x7f6d491454c4 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libFreeCADGui.so+0x134
#15  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Core.so.5(+0x2e9214) [0x7f6d468e9214]
#16  0x7f6d4776a422 in QAction::triggered(bool) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Widgets.so.5+0x32
#17  0x7f6d4776cdeb in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Widgets.so.5+0xab
#18  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Widgets.so.5(+0x25cb9a) [0x7f6d4785cb9a]
#19  0x7f6d4785ccd7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Widgets.so.5+0xe7
#20  0x7f6d47945a4a in QToolButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Widgets.so.5+0xa
#21  0x7f6d477aebd8 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Widgets.so.5+0x178
#22  0x7f6d4777035e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#23  0x7f6d47778222 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#24  0x7f6d490f7129 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libFreeCADGui.so+0x99
#25  0x7f6d468b3fc8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Core.so.5+0x108
#26  0x7f6d477763ab in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#27  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e39) [0x7f6d477c6e39]
#28  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Widgets.so.5(+0x1c9f99) [0x7f6d477c9f99]
#29  0x7f6d4777035e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#30  0x7f6d490f7129 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libFreeCADGui.so+0x99
#31  0x7f6d468b3fc8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Core.so.5+0x108
#32  0x7f6d46f34bec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Gui.so.5+0x84c
#33  0x7f6d46f0ad7c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Gui.so.5+0xac
#34  /tmp/.mount_FreeCAdAUZOv/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x6454a) [0x7f6d413fc54a]
#35  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/./libglib-2.0.so.0(+0x5a417) [0x7f6d44d11417]
#36  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/./libglib-2.0.so.0(+0x5d3c7) [0x7f6d44d143c7]
#37  /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f6d44d149be]
#38  0x7f6d469083d6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Core.so.5+0x66
#39  0x7f6d468b2bbb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Core.so.5+0x11b
#40  0x7f6d468ba6a2 in QCoreApplication::exec() from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libQt5Core.so.5+0x82
#41  0x7f6d4905bddc in Gui::Application::runApplication() from /tmp/.mount_FreeCAdAUZOv/usr/bin/../lib/libFreeCADGui.so+0x1ecc
#42  /tmp/.mount_FreeCAdAUZOv/usr/bin/freecad(+0x7620) [0x55c400e73620]
#43  /lib/x86_64-linux-gnu/libc.so.6(+0x276ca) [0x7f6d464456ca]
#44  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f6d46445785]
#45  /tmp/.mount_FreeCAdAUZOv/usr/bin/freecad(+0x7981) [0x55c400e73981]

```

### Full version info

```shell
[code]
OS: Debian GNU/Linux trixie/sid (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-12:
Please add:
- steps to reproduce this bug
- demo file / screenshots, anything 
- which workbench and command were you using?

Could you try if it still crashes with the dev builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

maxwxyz, 2024-06-15:
Closed for no feedback

closed 2024-06-15 by maxwxyz