FreeCAD/FreeCAD #13980: Sketch color doesn’t change with theme

pierreporte, 2024-05-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When a sketch is created and viewed in the 3D view from outside sketch mode, it has a color defined by the theme: white with classic and OpenDark, black with OpenLight. When the theme is changed, the color stays the same: a sketch created with the classic theme will appear white if the theme is changed to OpenLight, making it barely visible against the light grey background.

The color of sketch shouldn’t be named directly, similar to word processor that have an “automatic” color, that can change depending on other parameters. FreeCAD should do the same, with an automatic sketch color, which value is set by the theme and not hard-coded in the theme. The color becomes a sketch parameter once the user changes it directly. They could even select back the automatic color if it was already changed.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37249 (Git) AppImage
Build type: Release
Branch: main
Hash: 3acbc561f299b4cddea8215c387a52f3d5132cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * BillOfMaterials-WB 0.0.7
  * CosmeticThread3D
  * OpenTheme 2024.4.20
  * Curves 0.6.35
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-12:
@MisterMakerNL FYI

MisterMakerNL, 2024-05-12:
Can't wait till this is fixed!
I mean a lot has been fixed but this one is one hell of an annoying thing!
If someone needs me to test anything I'm here!

PaddleStroke, 2024-05-15:
I agree that this is very annoying.
The same is true for solid colors. The default color is saved in the document object. So changing theme does not change the default color.

PaddleStroke, 2024-05-15:
[image]
It is to be noted that the properties for the colors of point and lines are the same for sketch and other solids.

So if we want to have default colors different for solids and sketch, as it currently is, then we need to be smart about it.

Perhaps we could have a enum property "Colors" = [ default1, default2, custom]
If the property is set to custom, then it uses the properties "Line color" and "Point color". Else it uses colors stored in preference.

PaddleStroke, 2024-05-15:
Alternatively we choose 2 colors to refer to default. So if you set the point color to [0,1,2] then instead of a almost-black, we detect that its this particular color, and then we change it to default color 1 from preference.

We could have several default colors by using near-black values : 
default1 : [0,0,1]
default2 : [0,1,0]
default3 : [1,0,0]
default4 : [0,1,1]
and so on.

This has the advantage of being more flexible than previous solution : each color can be default or not. And also it does not need us to add more properties.

It would not be backward compatible with existing documents. But I don't think any solution could be...

MisterMakerNL, 2024-05-15:
Couldn't we add 2 colors to sketch, one is the version that is taken from the settings.
The other is a version that is taken from the file.
The file version overrules the settings version.
The file version only gets set when the user sets it.
The file version will be the current version.
The settings version will be new.
IF we could make a macro to reset old files then someone who is working on old files can reset it if they want.

PaddleStroke, 2024-05-20:
@wwmayer can you please weigh in what is your prefered solution for this?

MisterMakerNL, 2024-05-23:
This thing stops me from doing light backgrounds...

PR #14322 (2024-06-03): Gui: Update sketcher colors after file load
This ensures that color of sketch gets updated with user changing preferences and after loading files. PointColor and LineColor should be hidden, marked as transient and removed from existing documents. This should still work for older versions of FreeCAD as it will load defaults.

To ensure backward compatiblity migration strategy is to compute default value for this property based on the color. If it is non-white (the default value in FreeCAD) we assume that the change was inentional and that it should be kept as-is turning automatic coloring off.  [... cut]
closed 2024-06-03 by chennes