FreeCAD/FreeCAD #13981: Hidden selection is invisible

pierreporte, 2024-05-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When something is selected in FreeCAD (solid, edge face, …) and the view is rotated so that the selection is hidden behind geometry, the selection is invisible. It is impossible to know that something has been selected. It is then hard for the user to identify exactly what has been selected for an operation, especially complex ones.

FreeCAD could always display the edges of the selection so that it stands out in the 3D view. To better indicate that the hidden part of the selection is behind existing objects, it could be dashed. The visible part of the selection would remain as it is today.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37249 (Git) AppImage
Build type: Release
Branch: main
Hash: 3acbc561f299b4cddea8215c387a52f3d5132cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * BillOfMaterials-WB 0.0.7
  * CosmeticThread3D
  * OpenTheme 2024.4.20
  * Curves 0.6.35
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Core

### Anything else?

Solidworks does it:

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-12:
Related: https://github.com/FreeCAD/FreeCAD/issues/11214

darrellenns, 2024-06-05:
Here's an [example screen capture](https://imgur.com/a/freecad-select-hidden-linkstage-vs-0-22dev-KbXnRWa) comparing behavior of linkstage to dev.

In linkstage, the transparency not only makes the current selection visible but also allows for easily selecting other obscured features. This significantly reduces the number of steps needed to make selections, and makes it much more intuitive.

maxwxyz, 2025-02-04:
@kadet1090 could this issue benefit from transparent preview changes (render outlines of selection always on top)?