FreeCAD/FreeCAD #13983: Problems building in Fedora 40, due to missing depenencies. (Specifically docker container, but probably Fedora in general.)

garrett, 2024-05-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tried to build FreeCAD for Fedora using the container method and encountered two problems:

1. The instructions for building using docker are incorrect. The file is at `tools/build/Docker/Dockerfile.Fedora` and not at `tools/build/Dockerfile.Fedora` (this looks like a problem for all container build instructions). Basically, the path is not correct in the documentation.

2. Dependencies are not met properly. This is likely a larger issue with building on Fedora 40 in general.

When trying to run in Podman, I immediately have unmet dependency issues:

```
~/Source/FreeCAD main ❯ podman build --file tools/build/Docker/Dockerfile.Fedora --tag freecad-fedora
STEP 1/5: FROM fedora:latest
Resolved "fedora" as an alias (/etc/containers/registries.conf.d/000-shortnames.conf)
Trying to pull registry.fedoraproject.org/fedora:latest...
Getting image source signatures
Copying blob 504d52d8acd9 done   | 
Copying config 697048ff3d done   | 
Writing manifest to image destination
STEP 2/5: COPY fedora.sh /tmp
--> dec7becbee42
STEP 3/5: RUN dnf update --assumeyes &&     sh /tmp/fedora.sh &&     mkdir /builds
Fedora 40 - x86_64                              7.2 MB/s |  20 MB     00:02    
Fedora 40 openh264 (From Cisco) - x86_64        641  B/s | 1.8 kB     00:02    
Fedora 40 - x86_64 - Updates                    339 kB/s | 5.7 MB     00:17    
Dependencies resolved.
Nothing to do.
Complete!
Last metadata expiration check: 0:00:02 ago on Sun May 12 14:39:57 2024.
No match for argument: pyside2-tools
Package python3-3.12.3-2.fc40.x86_64 is already installed.
No match for argument: python3-pyside2-devel
Error: Unable to find a match: pyside2-tools python3-pyside2-devel
Error: building at STEP "RUN dnf update --assumeyes &&     sh /tmp/fedora.sh &&     mkdir /builds": while running runtime: exit status 1
```

It looks like Fedora does not ship `pyside2-tools` (and `python3-pyside2-devel`).

### Full version info

```shell
Git main as of today, commit ec44573bdce2707bc296ae85d9b6295689fe2536.
```

### Subproject(s) affected?

None

### Anything else?

This is the the result of a discussion on Mastodon, where building FreeCAD on Fedora no longer works:
https://mastodon.xyz/@mcdanlj@social.makerforums.info/112425605686318235

I personally don't use FreeCAD myself, but figured I'd spend a couple minutes seeing if building in a container would work before recommending it.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

johnsonm, 2024-05-12:
Pyside2 was dropped from Fedora a few releases ago, I think due to not keeping up with Python3 versions. I expect this build not to function right until FreeCAD moves to Pyside6 and Qt6, whenever that is.

howie-j, 2024-05-12:
Correct, fedora 38 was the last version with all dependencies necessary to build freecad. I use a toolbox with fedora 38 to build and run freecad, works perfectly: https://github.com/howie-j/freecad-development-setup/tree/main

johnsonm, 2024-05-12:
In a debian container, I tried using the packages from the docker build script tools/build/Docker/debian.sh

```
apt-get install --no-install-recommends --yes build-essential cmake doxygen \
    git libboost-date-time-dev libboost-dev libboost-filesystem-dev \
    libboost-graph-dev libboost-iostreams-dev libboost-program-options-dev \
    libboost-python-dev libboost-regex-dev libboost-serialization-dev \
    libboost-thread-dev libcoin-dev libeigen3-dev libgtkglext1-dev libgts-dev \
    libkdtree++-dev libkml-dev libmedc-dev libocct-data-exchange-dev \
    libocct-draw-dev libocct-foundation-dev libocct-modeling-algorithms-dev \
    libocct-modeling-data-dev libocct-ocaf-dev libocct-visualization-dev \
    libopencv-dev libproj-dev libpyside2-dev libqt5svg5-dev \
    libshiboken2-dev libvtk9-dev libvtk9-qt-dev \
    libvtk-dicom-dev libx11-dev libxerces-c-dev libxmu-dev libxmuu-dev \
    libzipios++-dev netgen netgen-headers pyside2-tools python3-dev \
    python3-matplotlib python3-pivy python3-ply python3-pyside2.qtsvg \
    python3-pyside2.qtuitools qtchooser qttools5-dev shiboken2 swig
```

However, that missed libyaml-cpp-dev and libqt5xmlpatterns5-dev that are now required.

johnsonm, 2024-05-13:
I see that d51d2f674 removed libqt5xmlpattherns5-dev but the build failed with it removed, so I don't really know where that bug is exactly...

yorikvanhavre, 2024-05-14:
> In a debian container, I tried using the packages from the docker build script tools/build/Docker/debian.sh However, that missed libyaml-cpp-dev and libqt5xmlpatterns5-dev and python3-pyside2.qtnetwork that are now required.

The wiki instructions are up-to-date: https://wiki.freecad.org/Compile_on_Linux but indeed that build script needs to be updated.

johnsonm, 2024-05-15:
True but less helpful. ☺

Cutting and pasting from a long bullet list of package names, or worse, re-typing, is mind-numbing.

If the wiki instructions linked to those scripts and they were kept up to date as the primary way of listing the packages, it would sure be easier to use...

luzpaz, 2024-07-21:
Any traction here? Would be great to clean this up before 1.0 release

hyarion, 2025-06-10:
> Pyside2 was dropped from Fedora a few releases ago, I think due to not keeping up with Python3 versions. I expect this build not to function right until FreeCAD moves to Pyside6 and Qt6, whenever that is.

We're on qt6 now.
Is this still an issue?

johnsonm, 2025-08-10:
@garrett given that I'm building regularly in a Fedora 41 container for much of this year, and the old dependencies have been dealt with, I suggest closing this one.