FreeCAD/FreeCAD #13989: Sketcher geometry not visible on the existing 3D geometry with default colors

FEA-eng, 2024-05-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This was pointed out by @chrisb-github [on the forum](https://forum.freecad.org/viewtopic.php?t=86738&start=10#p752291) but I haven't found an issue for it. With the current defaults, the sketcher geometry is not visible when a sketch is created on the existing 3D geometry:

[image]

[image]

I think that it would be good to fix this for 1.0.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-05-13:
> I think that it would be good to fix this for 1.0.

Surely releasing before this is sorted would be a serious mistake, it'll cause way too many reports on the forum. At the very least the shininess of existing objects should be reduced while in Sketcher Edit Mode and put back on exiting Edit Mode. I personally would be happy for the shininess to be reduced overall but apparently the DWG have ruled already.

FEA-eng, 2024-05-13:
> Surely releasing before this is sorted would be a serious mistake, it'll cause way too many reports on the forum. At the very least the shininess of existing objects should be reduced while in [...]

Exactly. This issue should be added to the 1.0 release project with must have fixes once it's created. Shininess reduction or headlight direction change (suggested on the forum) seems to be the best solution ATM.

maxwxyz, 2024-05-13:
This is realated to the specularity of the underlying geometry. With no object present it's visible (depending on the background color the user has chosen)
 https://github.com/FreeCAD/FreeCAD/issues/11166#issuecomment-1828460603

FEA-eng, 2024-05-13:
> This is related to the specularity of the underlying geometry. With no object present it's visible (depending on the background color the user has chosen) [#11166 (comment)](https://github.com/FreeCAD/FreeCAD/issues/11166#issuecomment-1828460603)

Ok but there's no separate issue to change this annoying default behavior in whatever way for the upcoming release?

Syres916, 2024-05-13:
@maxwxyz I know this subject is your own implementation but in this instance IMHO you've overstepped in closing this so quickly. If there was an assigned developer to fix the default use for a new user then I could understand but I can't see any such assignee. It would be a fudge for the user to have to keep moving the light direction when they are modelling pitched roofs for instance.

maxwxyz, 2024-05-13:
This is just related to the object becoming white when viewing a plane from their normal direction.
If you have no geometry, other geometry (e.g. a shpere) or a body with different material (color, shininess) it is perfectly visible.
I think changing a default line color is just hacking around the symptoms but the root cause should be fixed instead.

FEA-eng, 2024-05-13:
> This is just related to the object becoming white when viewing a plane from their normal direction. If you have no geometry, other geometry (e.g. a shpere) or a body with different material (color, [...]

Yes, the default sketcher geometry color is not what should be changed in this case and it doesn't always happen but it happens with the default settings for the simplest possible scenario like the one presented in the issue description. Thus, I think that it's very important to fix this and I would keep the issue open so that it's not forgotten. Maybe the easiest way would be to turn off or modify the lighting when entering a sketch. There could be a preference for this.

Syres916, 2024-05-14:
Two suggestions added to the forum thread https://forum.freecad.org/viewtopic.php?t=86738&start=40 , personally the headlight auto dipping in Sketch Edit Mode is my preferred one.

FEA-eng, 2024-05-14:
@Syres916 Thanks, I also think that the second solution is the way to go. Can you send a PR?

Syres916, 2024-05-14:
>  I also think that the second solution is the way to go. Can you send a PR?

I can but could @obelisk79 please give a view on https://forum.freecad.org/viewtopic.php?p=759295#p759295 for the auto dipping of the headlight intensity in Sketcher Edit Mode first.

obelisk79, 2024-05-14:
@Syres916 I engaged the DWG on your proposed solution and while there was some back and forth about what would work best no one disagreed about this being effective and simple. I think you should go ahead and generate the PR.

maxwxyz, 2024-05-14:
If it's easier, maybe disable specular color globally when a sketch is active?

maxwxyz, 2024-05-28:
Now we have the problem with light themes:
[image]
Shall I reopen this issue?

maxwxyz, 2024-05-28:
I'll reopen this as also images which you want to trace are reduced in visibility now with #14013 merged.

maxwxyz, 2024-06-08:
another example of the issue with the current implementation: I need to select this external edge and I need to rotate the model to even select it, but it is so dark, I cannot see the line:
[image]

MisterMakerNL, 2024-06-08:
> another example of the issue with the current implementation: I need to select this external edge and I need to rotate the model to even select it, but it is so dark, I cannot see the line: ![grafik](https://private-user-images.githubusercontent.com/6246609/337861502-6a879256-5ae7-4445-a9e5-d933ee85a309.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc4NDE1MjYsIm5iZiI6MTcxNzg0MTIyNiwicGF0aCI6Ii82MjQ2NjA5LzMzNzg2MTUwMi02YTg3OTI1Ni01YWU3LTQ0NDUtYTllNS1kOTMzZWU4NWEzMDkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDhUMTAwNzA2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2E4ZTJmZWU0NmU3M2ZjOTUzZjc0MTI5MGRhMjYwN2ZjNWIzZTczOGRkMGQ1NjE5MmQ4YzliY2JmZjkyNzU2YSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.8E4wyp2SuSIeK2HYTTWI29s3M9LeZCyNBiId549qM_E)

This tells me we need at least 3 point lights