FreeCAD/FreeCAD #13991: Start: clicking on tet on new file tiles does nothing

pierreporte, 2024-05-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On the new start page, when I click on the text on the new file tiles, nothing happens. To create a new file, I need to click on the empty space or on the icon. On the picture below, I’ve drawn red rectangle on the zone where a click does nothing. The real boundary is hard to determine but it’s definitely linked to the text.

[image]

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37249 (Git) AppImage
Build type: Release
Branch: main
Hash: 3acbc561f299b4cddea8215c387a52f3d5132cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * BillOfMaterials-WB 0.0.7
  * CosmeticThread3D
  * OpenTheme 2024.4.20
  * Curves 0.6.35
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-05-13:
I can reproduce it. However, it's only easily doable on the bold text. If I try to click on the regular text (bigger red area in the issue description), I can always activate the button, thus I cannot confirm there.

Also, when reproducing this issue, the button gets highlighted with a green rectangle, instead of the blue one that appears when it's correctly activated.

yorikvanhavre, 2024-05-14:
For me these icons look like this. I thought it was still work in progress... It's a bit different with other themes but always a very tiny font

[image]

FEA-eng, 2024-05-15:
I can confirm this on Windows but only for the titles.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git)
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

Syres916, 2024-05-17:
See https://github.com/FreeCAD/FreeCAD/issues/14039#issuecomment-2118391013

Syres916, 2024-05-21:
@yorikvanhavre do you get a different (hopefully improved) result with https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Fix_New_User_Start_Buttons ?

pierreporte, 2024-08-13:
This has been solved by #15772. Tested with the following version:

```
OS: Ubuntu 24.04 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git) AppImage
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * CosmeticThread3D
  * Curves 0.6.39
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.19
```

closed 2024-08-13 by pierreporte