FreeCAD/FreeCAD #13993: Wrong export in STL (export with errors)

lninio, 2024-05-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Exports parts to STL format generates incorrect invisible geometry or errors. It was possible to identify incorrect geometry using the Cura 3d printing slicer when loading the resulting STL model. As a result of the export, after processing the STL file using the Cura slicer, an extensive cavity is formed in which the part is not filled. If you export the model in 3 mf format, then the incorrect geometry is not generated, the filling of the part in the Cura slicer occurs completely. Also, processing the stl file exported from freecad in Prusa Prusa Slicer 2.7.4 shows that the original stl file contains errors.

### Full version info

```shell
[code]
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * Silk 0.1.5
  * Render 2024.1.23
  * OpticsWorkbench 1.0.17
  * workfeature
  * woodworking 0.21.2.33771
  * pcb 6.2023.1
  * retr3d
  * ExplodedAssembly
  * LCInterlocking
  * Dracula 0.0.5
  * Assembly3 0.12.2
  * Assembly4 0.50.12
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * sheetmetal 0.4.13
  * fasteners 0.5.20
  * CurvedShapes 1.0.9
  * Help 1.0.3
  * POV-Ray-Rendering
  * kicadStepUpMod 11.1.3
  * A2plus 0.4.64a
  * fasteners.backup1714339405.094781 0.5.17 (Disabled)
  * Curves 0.6.35
  * Dracula.backup1712833970.3185592 0.0.4 (Disabled)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[Air-filter2_Kryshka_v2.zip](https://github.com/FreeCAD/FreeCAD/files/15299284/Air-filter2_Kryshka_v2.zip)
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-14:
Could this be related to https://github.com/FreeCAD/FreeCAD/issues/8760 as I get a few errors when opening the file?
Please try this behavior also on the dev builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
How do you export the STL? Just export as .STL or create a mesh with the MeshWB?

lninio, 2024-05-24:
Hello @maxwxyz!
On the question of the model method in the STL Export file, my answer: I export using the File -> Export command.
To the question of whether this is a relationship to the problem with the number #8760: perhaps my question was incorrectly understood. I have no complaints about how the geometric mesh looks like (although the model contains missing faces through which you can see another geometry). I ask you to pay attention to the ring cavity in 3D models, which is visible on the result of the Cura slicer. This ring cavity is absent in the initial geometry, also this ring cavity does not appear in the Cura slicer when exporting to 3MF format. If I upload the exported STL file to the PRUSA slicer, then he reports errors in the geometry and offers to fix the model. After that, I can save the STL file in the PRUSA slicer and import into the Cura slicer again. After this action, I do not observe the ring cavity in the Cura slicer.
Based on the described above, I assume that there is a problem of export of Freecad model to the STL format.
Unfortunately, I am not a programmer, and I can’t investigate why this happens. I can attach the source 3D model of the part for research so that one of the FreeCAD team programmers can make a qualified investigation.
Earlier, I attached a ZIP archive with all files that are intended to confirm my conclusions and should help in the investigation. The STL file with the "fixed" suffix in the title is a file saved in the Prusa slice after correcting the model errors.

luzpaz, 2024-09-11:
@lninio can you reproduce on 1.0RC  ? (https://blog.freecad.org/2024/09/10/the-first-release-candidate-of-freecad-1-0-is-out/)

lninio, 2024-11-12:
> @lninio can you reproduce on 1.0RC ? (https://blog.freecad.org/2024/09/10/the-first-release-candidate-of-freecad-1-0-is-out/)

[image]

Hi luzpaz! It is seems OK

closed 2024-11-13 by maxwxyz