FreeCAD/FreeCAD #13997: Addon Manager Locks up FreeCAD

obelisk79, 2024-05-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening the Addon Manager in the latest weekly build it locks up FreeCAD and a full application process termiantion is required.

Error messages from console when starting addon-manager.
`
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAAv5LSI/usr/Mod/AddonManager/addonmanager_workers_startup.py", line 450, in run
    repo = Addon.from_cache(item)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAAv5LSI/usr/Mod/AddonManager/Addon.py", line 337, in from_cache
    instance._load_installed_metadata()
  File "/tmp/.mount_FreeCAAv5LSI/usr/Mod/AddonManager/Addon.py", line 387, in _load_installed_metadata
QResource: error decompressing zlib content (-3)
QResource: error decompressing zlib content (-3)
    self.installed_metadata = MetadataReader.from_file(installed_metadata_path)
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAAv5LSI/usr/Mod/AddonManager/addonmanager_metadata.py", line 255, in from_file
    return MetadataReader.from_bytes(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAAv5LSI/usr/Mod/AddonManager/addonmanager_metadata.py", line 261, in from_bytes
    return MetadataReader._process_element_tree(element_tree)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAAv5LSI/usr/Mod/AddonManager/addonmanager_metadata.py", line 267, in _process_element_tree
    return MetadataReader._create_node(namespace, root)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAAv5LSI/usr/Mod/AddonManager/addonmanager_metadata.py", line 376, in _create_node
    MetadataReader._parse_child_element(namespace, content_child, new_content_item)
  File "/tmp/.mount_FreeCAAv5LSI/usr/Mod/AddonManager/addonmanager_metadata.py", line 303, in _parse_child_element
QResource: compressed content does not fit into a QByteArray; use QFile instead
QResource: compressed content does not fit into a QByteArray; use QFile instead
    metadata.license.append(MetadataReader._parse_license(child))
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAAv5LSI/usr/Mod/AddonManager/addonmanager_metadata.py", line 322, in _parse_license
    lm = get_license_manager()
         ^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAAv5LSI/usr/Mod/AddonManager/addonmanager_licenses.py", line 186, in get_license_manager
    _LICENSE_MANAGER = SPDXLicenseManager()
                       ^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAAv5LSI/usr/Mod/AddonManager/addonmanager_licenses.py", line 54, in __init__
    self._load_license_data()
  File "/tmp/.mount_FreeCAAv5LSI/usr/Mod/AddonManager/addonmanager_licenses.py", line 64, in _load_license_data
    raw_license_data = json.loads(string_data)
                       ^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAAv5LSI/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAAv5LSI/usr/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAAv5LSI/usr/lib/python3.11/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
`

### Full version info

```shell
OS: EndeavourOS (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git) AppImage
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
  * Render 2024.1.23
  * Curves 0.6.35
  * Dracula 0.0.5
  * TabBar
  * IconThemes
  * fasteners 0.5.17
  * BIM 2021.12.0
  * FreeCAD_Dark.scss
  * Design456 0.0.1
  * fasteners.backup1712610431.2042964 0.5.17 (Disabled)
  * sheetmetal 0.4.11
  * OpenTheme 2024.4.20
  * SelectorToolbar
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kinghat, 2024-05-14:
same:

```
$ ./FreeCAD_weekly-builds-37302-conda-Linux-x86_64-py311.AppImage 
FreeCAD 0.22.0, Libs: 0.22.0devR37302 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
QResource: error decompressing zlib content (-3)
QIODevice::read (QFile, ":/licenses/spdx.json"): device not open
Traceback (most recent call last):
  File "/tmp/.mount_FreeCATpidVe/usr/Mod/AddonManager/addonmanager_workers_startup.py", line 450, in run
    repo = Addon.from_cache(item)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCATpidVe/usr/Mod/AddonManager/Addon.py", line 337, in from_cache
    instance._load_installed_metadata()
  File "/tmp/.mount_FreeCATpidVe/usr/Mod/AddonManager/Addon.py", line 387, in _load_installed_metadata
    self.installed_metadata = MetadataReader.from_file(installed_metadata_path)
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCATpidVe/usr/Mod/AddonManager/addonmanager_metadata.py", line 255, in from_file
    return MetadataReader.from_bytes(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCATpidVe/usr/Mod/AddonManager/addonmanager_metadata.py", line 261, in from_bytes
    return MetadataReader._process_element_tree(element_tree)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCATpidVe/usr/Mod/AddonManager/addonmanager_metadata.py", line 267, in _process_element_tree
    return MetadataReader._create_node(namespace, root)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCATpidVe/usr/Mod/AddonManager/addonmanager_metadata.py", line 376, in _create_node
    MetadataReader._parse_child_element(namespace, content_child, new_content_item)
  File "/tmp/.mount_FreeCATpidVe/usr/Mod/AddonManager/addonmanager_metadata.py", line 303, in _parse_child_element
    metadata.license.append(MetadataReader._parse_license(child))
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCATpidVe/usr/Mod/AddonManager/addonmanager_metadata.py", line 322, in _parse_license
    lm = get_license_manager()
         ^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCATpidVe/usr/Mod/AddonManager/addonmanager_licenses.py", line 186, in get_license_manager
    _LICENSE_MANAGER = SPDXLicenseManager()
                       ^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCATpidVe/usr/Mod/AddonManager/addonmanager_licenses.py", line 54, in __init__
    self._load_license_data()
  File "/tmp/.mount_FreeCATpidVe/usr/Mod/AddonManager/addonmanager_licenses.py", line 64, in _load_license_data
    raw_license_data = json.loads(string_data)
                       ^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCATpidVe/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCATpidVe/usr/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCATpidVe/usr/lib/python3.11/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

```

> OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.37302 (Git) AppImage
> Build type: Release
> Branch: main
> Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
> Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
> Locale: English/United States (en_US)
> Installed mods: 
>   * Curves 0.6.36
>   * CurvedShapes 1.0.9
>   * OpenTheme 2024.5.3

maxwxyz, 2024-05-14:
can reproduce on Windows too.
@chennes FYI

krushia, 2024-05-15:
Confirmed. Build 37249 is fine.
I'd like to place a bet on https://github.com/FreeCAD/FreeCAD/commit/2e6af15880ad84b56f44cf99c343fa370b499b5b

chennes, 2024-05-15:
Are you all running AppImage/Conda builds? Any self-compilers reproducing this?

obelisk79, 2024-05-15:
> Are you all running AppImage/Conda builds? Any self-compilers reproducing this?

I've finally been able to build locally (Qt6) and the addon manager seems to work, so perhaps it's related to the automated build for the appimages?

kinghat, 2024-05-15:
> Are you all running AppImage/Conda builds? Any self-compilers reproducing this?

appimages here

marcuspollio, 2024-05-15:
> Any self-compilers reproducing this?

Local build:
 
```
OS: GNU/Linux Ubuntu 22.04 LTS based (DE:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37303 (Git)
Build type: Release
Branch: main
Hash: 869cb1f83b5f92ee134dafde46c9c5e0dceb8889
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.5.1
```

krushia, 2024-05-18:
Seems to be fixed now?

OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36
  * A2plus 0.4.64a
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * fasteners 0.5.20
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools
  * workfeature
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.5.3
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23

chennes, 2024-05-19:
Yes, I had to roll back an icon change, somehow it was corrupting the Qt resource file on some platforms. We're going to try again, with smaller change sets in between, to try to narrow down which icon is causing the problem.

maxwxyz, 2024-05-20:
I'll close this issue as the revert is merged. Let me know if the issue becomes relevant again.

closed 2024-05-20 by maxwxyz