FreeCAD/FreeCAD #13998: Gizmo mouse click events are not registered

dov, 2024-05-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When pressing the mouse button 1 on the gizmo in the 3d view, no actions are performed.

### What I did

1. Opened an arbitrary part in the FreeCAD and rotated it around in the 3D view.
2. Pressed the "Top" label of the Gizmo

### What happened

Nothing

### What I expected to happen

I expected to view the model in the Top view (negative z direction). This is what happens in my snap version of FreeCAD 0.21 .

### Additional info

Mouse presses on the arrows or the rotation circles in the Transform tools don't work either.

### Full version info

```shell
OS: Fedora Linux 40 (Workstation Edition) (fvwm)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37096 (Git)
Build type: Debug
Branch: main
Hash: af2b8b661f711f8512d63345e4ffe321e292ddbb
Python 3.12.2, Qt 6.7.0, Coin 4.0.2, Vtk 9.2.6, OCC 7.8.1.dev
Locale: C/Default (C)
Installed mods: 
  * Curves 0.6.21
  * MyWorkbench
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-14:
Semms to me like a duplicate of https://github.com/FreeCAD/FreeCAD/issues/13494 can you confirm?

dov, 2024-05-14:
@maxwxyz Indeed looks the same. I'll close this issue as a duplicate. Thanks.

Duplicate of #13494

closed 2024-05-14 by dov