FreeCAD/FreeCAD #13999: Sketcher segfaults when using simmetry

furgo16, 2024-05-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Uncompress and open attached file 
2. Double click on the `Sketch` sketch on the tree view
3. In the `Tasks` panel, under `Elements`, select the last four lines (`50-Line`, `51-Line`, `52-Line`, `53-Line`)
4. Click on the Sketcher simmetry command (`Sketcher_Simmetry`) on the toolbar
5. Move the mouse so as to position the selected lines on the right hand side of their original position, using the vertical construction line (`49-Line`) as the axis of simmetry
6. When the desired position is achieved, click Enter to confirm the operation
7. Expected: simmetry operation is applied
8. Actual: segmentation fault, FreeCAD crashes and exits

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37303 (Git) Snap 1019
Build type: Release
Branch: main
Hash: 869cb1f83b5f92ee134dafde46c9c5e0dceb8889
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
```

### Subproject(s) affected?

Sketcher

### Anything else?

[File to reproduce](https://github.com/FreeCAD/FreeCAD/files/15303452/windows_front_bug.zip)

Segmentation fault

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f4dc37e8520]
#1  0x7f4d81380675 in non-virtual thunk to SketcherGui::DrawSketchControllableHandler<SketcherGui::DrawSketchDefaultWidgetController<SketcherGui::DrawSketchHandlerSymmetry, SketcherGui::StateMachines::OneSeekEnd, 0, SketcherGui::OnViewParameters<0>, SketcherGui::WidgetParameters<0>, SketcherGui::WidgetCheckboxes<2>, SketcherGui::WidgetComboboxes<0>, SketcherGui::ConstructionMethods::DefaultConstructionMethod, false> >::onModeChanged() from /snap/freecad/1015/usr/lib/SketcherGui.so+0x75
#2  0x7f4d8135b572 in SketcherGui::DrawSketchControllableHandler<SketcherGui::DrawSketchDefaultWidgetController<SketcherGui::DrawSketchHandlerSymmetry, SketcherGui::StateMachines::OneSeekEnd, 0, SketcherGui::OnViewParameters<0>, SketcherGui::WidgetParameters<0>, SketcherGui::WidgetCheckboxes<2>, SketcherGui::WidgetComboboxes<0>, SketcherGui::ConstructionMethods::DefaultConstructionMethod, false> >::pressButton(Base::Vector2d) from /snap/freecad/1015/usr/lib/SketcherGui.so+0x92
#3  0x7f4d8145bffb in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) from /snap/freecad/1015/usr/lib/SketcherGui.so+0x22b
#4  0x7f4dc6ace32c in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /snap/freecad/1015/usr/lib/libFreeCADGui.so+0x15c
#5  0x7f4dc33adb47 in SoEventCallback::handleEvent(SoHandleEventAction*) from /snap/freecad/1015/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0xb7
#6  0x7f4dc31d6095 in SoAction::traverse(SoNode*) from /snap/freecad/1015/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x65
#7  0x7f4dc3370908 in SoChildList::traverse(SoAction*, int, int) from /snap/freecad/1015/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#8  0x7f4dc33b7660 in SoGroup::doAction(SoAction*) from /snap/freecad/1015/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x50
#9  0x7f4dc33cc66c in SoSeparator::doAction(SoAction*) from /snap/freecad/1015/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#10  0x7f4dc31d6095 in SoAction::traverse(SoNode*) from /snap/freecad/1015/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x65
#11  0x7f4dc3370908 in SoChildList::traverse(SoAction*, int, int) from /snap/freecad/1015/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#12  0x7f4dc33b7660 in SoGroup::doAction(SoAction*) from /snap/freecad/1015/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x50
#13  0x7f4dc33cc66c in SoSeparator::doAction(SoAction*) from /snap/freecad/1015/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#14  0x7f4dc31d6095 in SoAction::traverse(SoNode*) from /snap/freecad/1015/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x65
#15  0x7f4dc31dfa9b in SoHandleEventAction::beginTraversal(SoNode*) from /snap/freecad/1015/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0xab
#16  0x7f4dc31d66e8 in SoAction::apply(SoNode*) from /snap/freecad/1015/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0xc8
#17  0x7f4dc3385559 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /snap/freecad/1015/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x79
#18  0x7f4dc6a446b2 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /snap/freecad/1015/usr/lib/libFreeCADGui.so+0x22
#19  0x7f4dc6a52ff9 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /snap/freecad/1015/usr/lib/libFreeCADGui.so+0x59
#20  0x7f4dc6a5e751 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /snap/freecad/1015/usr/lib/libFreeCADGui.so+0x41
[... 12 lines cut ...]
#33  0x7f4dc4ab3e87 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /snap/freecad/1015/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#34  /snap/freecad/1015/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c91cb) [0x7f4dc4b0a1cb]
#35  /snap/freecad/1015/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cc525) [0x7f4dc4b0d525]
#36  0x7f4dc4aad763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/1015/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#37  0x7f4dc663afa8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/1015/usr/lib/libFreeCADGui.so+0x98
#38  0x7f4dc3f13e4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/1015/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#39  0x7f4dc433c1e7 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /snap/freecad/1015/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x687
#40  0x7f4dc430fdbc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1015/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#41  /snap/freecad/1015/kf5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/../../../libQt5XcbQpa.so.5(+0x76f5e) [0x7f4dbd907f5e]
#42  /snap/freecad/1015/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7f4dc19acd3b]
#43  /snap/freecad/1015/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab258) [0x7f4dc1a02258]
#44  /snap/freecad/1015/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f4dc19aa3e3]
#45  0x7f4dc3f6d338 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1015/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#46  0x7f4dc3f1276b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1015/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#47  0x7f4dc3f1ad14 in QCoreApplication::exec() from /snap/freecad/1015/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#48  0x7f4dc6575ae4 in Gui::Application::runApplication() from /snap/freecad/1015/usr/lib/libFreeCADGui.so+0x6c4
#49  /snap/freecad/1015/usr/bin/FreeCAD(+0xa709) [0x559417380709]
#50  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f4dc37cfd90]
#51  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f4dc37cfe40]
#52  /snap/freecad/1015/usr/bin/FreeCAD(+0xaac5) [0x559417380ac5]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-05-14:
> Move the mouse so as to position the selected lines on the right hand side of their original position, using the vertical construction line (49-Line) as the axis of simmetry

This is truly unclear. I don't know how to reproduce the bug. Could you record a video?

furgo16, 2024-05-14:
> This is truly unclear. I don't know how to reproduce the bug. Could you record a video?

It's literally grabbing those four lines and creating a symmetry across the construction line. I'll try to upload a video as soon as I can.

xtemp09, 2024-05-14:
I cannot reproduce.

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37303 (Git)
Build type: Debug
Branch: main
Hash: 869cb1f83b5f92ee134dafde46c9c5e0dceb8889
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

furgo16, 2024-05-14:
Thanks for testing. I can still reproduce it, unfortunately.

A workaround I've found to be able to continue working on the model is to do the symmetry one line at a time, instead of selecting the four lines at once. Doing that the crash is not reproduceable.

maxwxyz, 2024-05-14:
@PaddleStroke FYI

could this be related to https://github.com/FreeCAD/FreeCAD/issues/13943

wwmayer, 2024-05-14:
This issue is not related to #13915.

The crash occurs inside afterHandlerModeChanged() at line 463 because of a heap-use-after-free error.
The sketchHandler will be destroyed in ViewProviderSketch::deactivateHandler() but ViewProviderSketch::mouseButtonPressed() it's accessed without checking beforehand whether it's null.

wwmayer, 2024-05-14:
The issue is a bit more complex. The DrawSketchHandlerSymmetry purges itself after the click (by calling ViewProviderSketch::deactivateHandler()). So, it will be destroyed inside a member function but then still accesses a member variable which of course causes a segmentation fault.

The relevant stacktrace  after the click into the 3D view is:
1. ViewProviderSketch::mouseButtonPressed
2. DrawSketchControllableHandler::pressButton
3. DrawSketchDefaultHandler::onButtonPressed
4. StateMachine::moveToNextMode
5. StateMachine::setState
6. DrawSketchControllableHandler::onModeChanged
7. DrawSketchDefaultHandler::onModeChanged
8. DrawSketchDefaultHandler::finish
9. DrawSketchDefaultHandler::handleContinuousMode
10. ViewProviderSketch::purgeHandler
11. ViewProviderSketch::deactivateHandler
12. Assign null to the unique pointer that destroys the DrawSketchHandlerSymmetry

One of the calling instances of the stack is still DrawSketchControllableHandler::onModeChanged and after the self-destruction it accesses `toolWidgetManager.afterHandlerModeChanged();` which is undefined behaviour.

PaddleStroke, 2024-05-15:
Do you have a solution in mind?

Is it possible to have `DSDefaultHandler::onModeChanged` returning a bool (false if handler is purged) and do like this :

```
    void onModeChanged() override
    {
        DrawSketchHandler::resetPositionText();
        toolWidgetManager.onHandlerModeChanged();
        if(DSDefaultHandler::onModeChanged()) {
               toolWidgetManager.afterHandlerModeChanged();
        }
    }
```

wwmayer, 2024-05-15:
The logic of the sketch handlers is somewhat convoluted because the view provider has a pointer to it and the handler has a pointer to the sketch. Then, the handler passes itself as argument to one of its members. This makes it difficult to understand the whole ownership.

> Is it possible to have DSDefaultHandler::onModeChanged returning a bool (false if handler is purged) and do like this :

Should work.

> Do you have a solution in mind?

I thought about removing the line to destroy the handler inside deactivateHandler() but set a flag that it's deactivated. Then the uppermost calling instance (in this case ViewProviderSketch::mouseButtonPressed) checks if the handler is deactivated and destroys it. However, this may require some more changes to make it working for every handler type.

PR #14029 (2024-05-17): Sketcher: fixes #13999
fixes #13999
@wwmayer I could not reproduce the crash on windows. Can you please check and feedback if this solves correctly the crash ?
commit c43658ea (2024-05-17): Sketcher: fixes #13999
closed 2024-05-17 by wwmayer